ESPresense/ESPresense #617: How do i find the Apple iPhone model when i want to set a filter. I have iPhone 10. Bur how Can i find about the model is 10-3 og 4 or 5 .......

Mindkey75, 2022-08-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

janssenjones, 2022-08-30:
This answer is not ideal, and things are still a bit in flux and changing.  You can find out a bit more info here:
https://espresense.com/configuration/settings#querying 

But in short, you basically want to go the configuration page and enter that string into the text box that says:
"Query device ids for characteristics (eg. apple:1005:9-26):"

And then save and restart the devices.  It'll then start making a secondary query that reports a new name (e.g. apple:watch6-4).

You may still get overlap, depending on if there are multiple running instances of iPhones or iPads or watches or whatever in your house.  A new system is coming to uniquely capture each device (using irk), but that's in the beta versions, and is still a bit finnicky.

More docs should be on the way soon for some of those things.

Mindkey75, 2022-08-30:
> Docs are now released and refactored. v3 is out. You need to press the Enroll button and connect to it via the Bluetooth screen and we should be able to track it uniquely.

Thank you it works perfect now👍🏻

 Mvh Morten Henriksen

closed 2022-08-30 by DTTerastar