ESPresense/ESPresense #619: Update by uploading firmware via web ui

janssenjones, 2022-08-28:
**Is your feature request related to a problem? Please describe.**
My ESP devices are all on a private network that don't have internet access.  Therefore, the auto-update feature doesn't work (which is OK with me).  

**Describe the solution you'd like**
I'd like to be able to either point the ESP devices to an internal URL for firmware or have a dialog box where I can provide a binary file (similar to how it's done via Tasmota or ESPHome)

If there's a way to update these devices without putting them on the internet or plugging them back in via USB, I'd be fine with any option.  I'm just not sure how to do it at the moment other than those two options, which are both less than idea.

onlyoneme, 2022-08-28:
There is a way to update remotely using Arduino protocol - "Arduino OTA Update" in the device settings. Never used it personally.

onlyoneme, 2022-08-28:
Just tried to OTA update with espota.py script. Works well.

janssenjones, 2022-08-29:
> Just tried to OTA update with espota.py script. Works well.

Thanks for testing and confirming.  I only had a few minutes to try today, but wasn't able to get it to work.  I set the Arduino OTA checkbox, and saved and rebooted, and then tried this from the terminal:
.\espota.py -i 192.168.2.60 -f .\esp32.bin
and
.\espota.py -i 192.168.2.60 -f .\esp32.bin -p 80

For both, I just got:
16:59:59 [ERROR]: No Answer

I did a port scan and the ESP is only listening on port 80, and I can browse the page, but the espota.py doesn't seem to do anything.  Could you share your syntax?

onlyoneme, 2022-08-29:
Did you enable Arduino OTA Update setting before? The script must connect to port 3232 on esp device, so in my case it looks like 
espota.py -d -i 192.168.205.73  -p 3232 -P 3232 -f esp32.bin
I use both ports because my esp device is on VLAN so I have to enable incoming traffic from device to my computer on router's firewall.
I've tested it with b8 and with downgrading to 2.2.9.

janssenjones, 2022-08-29:
Hmmm... I've confirmed the Arduino OTA checkbox is checked, and rebooted (and even removed and re-added power), but it's not listening on port 3232.  It only listens on 80:

`~$ nmap 192.168.2.60 -p 3232
Starting Nmap 7.80 ( https://nmap.org ) at 2022-08-29 12:38 UTC
Nmap scan report for 192.168.2.60
Host is up (0.097s latency).

PORT     STATE  SERVICE
3232/tcp closed mdtp

Nmap done: 1 IP address (1 host up) scanned in 0.15 seconds
~$ nmap 192.168.2.60 -p 80
Starting Nmap 7.80 ( https://nmap.org ) at 2022-08-29 12:38 UTC
Nmap scan report for 192.168.2.60
Host is up (0.097s latency).

PORT   STATE SERVICE
80/tcp open  http`

I guess I'll try flashing to a previous version and testing again.

onlyoneme, 2022-08-29:
I don't see this port opened during scanning, but update itself works. Just flashed from 2.2.9 to 2.2.9 once again.

onlyoneme, 2022-08-29:
Do you use esp32 version of espota.py?

janssenjones, 2022-08-29:
@onlyoneme  - Thanks for all your help on this one.  With the -d debug flag, I was able to get a better idea of what was going on.  On all my other setups (ESPHOME, Tasmota, etc), I believe I actually upload the file to the ESP, but in this case, it's sending a request for the ESP to download the file instead, and my networking is setup to not allow the ESP devices to get on my main network (it's a dedicated "device VLAN").  I just tried putting an ESP on my main WiFi network, and it flashed without issue using espota.py.  

I think my initial feature request remains - I'd like to be able to send files to the ESP without having to open up routes from the ESP to the internet or to my home network.

janssenjones, 2022-08-29:
> > it's sending a request for the ESP to download the file instead, and my networking is setup to not allow the ESP devices to get on my main network (it's a dedicated "device VLAN")
> 
> That's why I define outbound port as well to be able to create a rule for specified port on my computer. I can simply enable it on demand.

Got it.  Thanks again for your help!

DTTerastar, 2022-08-29:
What I think would be super cool is take ESPHome and strip out 90% of it and just keep the listing of the nodes and a button to espota a particular version to a node.

janssenjones, 2022-08-30:
I went ahead and followed the guidance and created the custom firewall rule that opens up the Device Vnet for a custom espota port that can be enabled on the fly.  Flashing all six devices was definitely faster than bringing them all back via USB cable, and was probably comparable to a web UI, so I think that's sufficient for me. I'll go ahead and close this request, as I'm guessing it's probably a pretty niche scenario.

closed 2022-08-30 by janssenjones