ESPresense/ESPresense #623: Devices with IRK registrations sometimes change how they report

janssenjones, 2022-08-29:
**Describe the bug**

I have several Apple devices with IRK registrations (I'm pretty much trying to just use these going forward, due to inconsistency with other methods), and they seem to inconsistently report between irk:myirkstring and MyPhone12.  I see this behavior both on the Devices#/Fingerprints screen and within MQTT. It seems to be the case that immediately after an ESP reboot, the phone shows up as irk:myirkstring, but if I disable and re-enable bluetooth on the phone, or leave range and come back, it then starts reporting, both on the ESP web page and in MQTT as MyPhone12.   Within Home Assistant, I believe I can only set one device_id, but it's currently getting a mix of both IRK strings and device names. 

At any given time, a device might be reporting to MQTT for one room via IRK, but another via devicename.  All ESP configs have identical "Known BLE identity resolving keys" strings.  It solely seems to be determined by if the device was present when the ESP rebooted.

**Config**

- Version: 3.0 b10 and 3.0 b11
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor:
- Device (be specific): D1 Mini ESP32

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

DTTerastar, 2022-08-30:
Load up MQTT explorer and look at espresense/settings  is there anything in there that indicates what's going on?

janssenjones, 2022-08-30:
I think I've got it root caused.  It's happening on ESP reboot, and it's happening because a subset of devices may often be detected before the code gets to the IRK mapping table.  This is what I see in the terminal logs:

```
ArduinoOTA configured and ready
1 Motion| Pir: no Radar: no
1 New   | MAC: a4c1....                                              -72dBm 
1 New   | MAC: 12316.....    -47dBm 
1 New   | MAC: 76b2....  irk:c7f7fe7b26 ....... -59dBm 071df5059058
etc...
1 MQTT  | Config irk:e1a.... {"id":"J12"}
0 MQTT  | Config irk:c7f7f...... {"id":"JPro"}
```

If the phone/iPad/etc connects to the ESP and reports before the mapping table gets initialized, then it keeps reporting as the IRK instead of the device name.  On each reboot, different devices seem to connect before the table initializes.

RaoulSargent, 2022-08-30:
Although I have not debugged (as per prev post) - I do see the same behaviour, that 'some' devices report as 'irk' rather than the friendly name assigned to them; and this behaviour appears random.
The scenario described in prev post describes my symptoms.

DTTerastar, 2022-08-30:
There is a race between mac and mqtt, but I tell every fingerprint to "check again" each time we get the mqtt config: https://github.com/ESPresense/ESPresense/blob/0fadc5b7bc579e258e003fdfe783b86c5b13f483/lib/BleFingerprint/BleFingerprintCollection.cpp#L81

The delay between the irk and the aliases name should be less than a couple seconds.

DTTerastar, 2022-08-30:
At the VERY most it should only take 15 minutes for an iPhone since they rotate their MACs that often.  Super strange.  Is anyone else seeing this issue?

janssenjones, 2022-08-30:
> At the VERY most it should only take 15 minutes for an iPhone since they rotate their MACs that often. Super strange. Is anyone else seeing this issue?

It's possible that it resolves after 15 minutes.  I'll set some stopwatches to try to confirm.  I've been making constant changes to BLE resolving keys, MQTT filters, distance to report etc, to try to dial things in and clear out noise, but from what I can tell, those changes all require a restart to take effect, so given the constant change, I'm getting as many IRK posts to MQTT as I am device names.  After everything's dialed in, maybe this will all become less noticeable and will work itself out...

janssenjones, 2022-08-30:
I just completed a restart with a stopwatch going, and it seemed to take 27 minutes for the iPad MAC to rollover and then get reported properly to MQTT.  Maybe they have different behavior based on the make/model/release year/OS version/et cetera.  I'm not sure.

DTTerastar, 2022-09-01:
I know what it is.  You're jamming your irk into known irks even tho it's also in mqtt, right?

janssenjones, 2022-09-01:
> I know what it is. You're jamming your irk into known irks even tho it's also in mqtt, right?

Are you saying that if you use the device enrollment method, then you're not supposed to populate Known BLE identity resolving keys?  I was thinking the enrollment method was just step 1 in that process rather than a replacement process.  I can clear all of those and re-test.

DTTerastar, 2022-09-01:
yep, can you fix the documentation as well?  Others are doing too!

DTTerastar, 2022-09-01:
I can make a new alias coming it recalc the ids as well, that's the issue.

janssenjones, 2022-09-01:
I just removed my IRKs and restarted, and everything came back instantly and perfectly.   So yep, that was the issue.  Thanks! I can try to work on updated docs tomorrow.

closed 2022-09-01 by DTTerastar

DTTerastar, 2022-09-01:
With v3.0.3 you can jam it into both and it'll work fine...

commit e9bb853a (2022-09-01): Fix when device config alias comes in late
Add support for device config rssi@1m and name
Fixes #623
PR #638 (2022-09-01): Fix when device config alias comes in late
Add support for device config rssi@1m and name
Fixes #623
commit 11a61d20 (2022-09-01): Fix when device config alias comes in late (#638)

Add support for device config rssi@1m and name
Fixes #623