ESPresense/ESPresense #627: ESP32 - Blue LED

bigal2112, 2022-08-30:
Around 4am this morning the blue LED on my ESP32's came on. The red power one has always been on and, although annoying, it's bearable but the blue one lights up the whole room!!

After looking at recent updates this may be to do with https://github.com/ESPresense/ESPresense/pull/595.

I've tried changing all the LED pins to -1 in the device calibration but this makes no difference.

Can this be turned off programmatically or is it time to get the pliers out??

nalkinburgh, 2022-08-30:
This is definitely something new with the latest update. I was able to get one to turn off by going back to the last 2.x.x firmware and then manually upgrading to the latest, with the setting at -1. Unfortunately none of mine will connect to MQTT this morning and after the initial install I can't even access them via wifi.

DTTerastar, 2022-08-30:
If all pins are -1 we shouldn't turn anything on or off.  But an ON neoPixel led will stay on until power is cut.  So try unplugging it?
What type of device?  And yes v3 has loads of new stuff.  Need separate issues please.

bigal2112, 2022-08-30:
Thanks DTTerastar that done the trick. I'm sure I tried that earlier this morning and it didn't work, but all good now. Just in case anyone else has the same issue the board was an ESP32 DEVKITV1 and the process I did was:

1. In the configuration set the LED 1 pin to -1
2. Save
3. Restart the device from within the configuration (probably not needed but hey ho)
4. Once rebooted I then unplugged it, gave it a couple of seconds then plugged it back in.

No more blue LED.

Thanks again for this and all your work on this project.

closed 2022-08-30 by DTTerastar