ESPresense/ESPresense #628: Latest release will not connect

nalkinburgh, 2022-08-30:
Starting last night when my devices updated to 3.0 then 3.0.1 they have been unable to connect to MQTT. I am using ESP32-D0WDQ6 development boards. I do have one D1 Mini that seems to be intermittently connecting but not consistent. I was able to downgrade one to 2.2.9 and it connected but all my sensors were gone and the "sesnor" firmware option seems to be gone on the install page.

Log Info that seems to be pertinent:
0 Reconnecting to MQTT...
0 Reconnect timer
Too many reconnect attempts; Restarting
Disconnected from MQTT; reason 0
ets Jun  8 2016 00:22:57

0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
1 New   | MAC: 4dd0795af5d2, ID: 4dd0795af5d2                                              -89dBm 
0 Reconnect timer
0 Reconnecting to MQTT...
1 New   | MAC: 695032e79316, ID: apple:1006:10-12                                          -91dBm 171e9bdbb1b7
Disconnected from MQTT; reason 0

abort() was called at PC 0x401b9f5e on core 1

Backtrace:0x40083d19:0x3ffe28300x40095fed:0x3ffe2850 0x4009b389:0x3ffe2870 0x401b9f5e:0x3ffe28f0 0x401b7a17:0x3ffe2910 0x4019f77e:0x3ffe2930 0x4019f74d:0x3ffe2950 0x40114401:0x3ffe2970 0x40114449:0x3ffe2990 0x400e51d4:0x3ffe29b0 0x400e51fc:0x3ffe29d0 0x401d2d1e:0x3ffe29f0 0x400e8b78:0x3ffe2a10 0x400e8bc9:0x3ffe2aa0 0x400e64a1:0x3ffe2ac0 0x400e4ca1:0x3ffe2b10 0x400e4d71:0x3ffe2b60 0x400e4f29:0x3ffe2bb0 0x401ba651:0x3ffe2bd0 0x401ba6d1:0x3ffe2c00 0x401bacd6:0x3ffe2c20 

ELF file SHA256: 0000000000000000

DTTerastar, 2022-08-30:
I've made all v3 firmwares include sensor.  That log looks like what I see when I have multiple nodes with the same room name.  It's because mqtt server disconnects one of the nodes when it sees duplicate client ids.

nalkinburgh, 2022-08-30:
They were all working last night on 2.2.9 and they all have different names. Regarding the sensor version, I was referring to 2.2.9 because I can downgrade and get it to connect, but all my sensor information was removed when I downgraded even though I told it not to erase the device. 

Here are the recent MQTT logs. You can see they have different names.

------
2022-08-30 10:10:57: New client connected from 10.0.0.249:49471 as espresense-diningroom (p2, c1, k15, u'mqtt').
2022-08-30 10:10:59: New connection from 172.30.32.2:56396 on port 1883.
2022-08-30 10:10:59: Client <unknown> closed its connection.
2022-08-30 10:11:43: Client espresense-diningroom has exceeded timeout, disconnecting.
2022-08-30 10:11:58: New connection from 10.0.0.124:56010 on port 1883.
2022-08-30 10:11:58: New client connected from 10.0.0.124:56010 as espresense-master (p2, c1, k15, u'mqtt').
2022-08-30 10:12:19: New connection from 10.0.0.249:49473 on port 1883.
2022-08-30 10:12:25: Client espresense-master has exceeded timeout, disconnecting.
2022-08-30 10:12:59: New connection from 172.30.32.2:56514 on port 1883.

DTTerastar, 2022-08-30:
More serial log would be helpful.  Is telemetry turned off?  We should be sending data every 15 seconds and shouldn't ever timeout.

el97, 2022-08-30:
I have similar issues. I get the same in the log output and I lose connection to the controller as well. It works fine on 2.2.9 though. If you need any more info please tell me how I can contribute.

nalkinburgh, 2022-08-31:
Here is more log. Telemetry is not turned off. These worked fine on 2.2.9 and they updated automatically. This one does not have any additional sensors attached.

Connected 115200 8N1...
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Firmware:     esp32
Version:      v3.0.2
WiFi Scan:    All (picked channel=7 bss=28:EE:52:A7:C2:23 rssi=-56)
IP address:   10.0.0.234
DNS address:  10.0.0.1
Hostname:     espresense-kitchen
Room:         kitchen
Mqtt server:  10.0.0.233:1883
Max Distance: 3.00
Init Free Mem:139100
PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    

1 Motion| Pir: no Radar: no
1 New   | MAC: 0aeb11efe478, ID: msft:cdp:0902                                             -84dBm 
1 New   | MAC: d93932353e41, ID: name:gvh5101_3e41                                         -68dBm 
1 New   | MAC: a4c138ddb14c, ID: a4c138ddb14c                                              -85dBm 
1 New   | MAC: a4c138d1e55d, ID: a4c138d1e55d                                              -73dBm 
1 New   | MAC: 53cac2843104, ID: apple:1005:9-24                                           -50dBm 049811e5fb
1 New   | MAC: 7005582586e9, ID: apple:1007:11-8                                           -43dBm 201fbbe94f3478
1 New   | MAC: f80dac927b75, ID: f80dac927b75                                              -60dBm 
1 New   | MAC: 67cb54357176, ID: apple:1005:9-12                                           -78dBm 0f1c0c2bbf
1 New   | MAC: 5a87807b25ff, ID: apple:1006:10-12                                          -86dBm 161e62812ab9
1 New   | MAC: 4844f4b79b74, ID: apple:1005:9-24                                           -75dBm 0118feab1b
1 New   | MAC: d33932352d41, ID: name:gvh5101_2d41                                         -64dBm 
1 New   | MAC: 6426dd0648dd, ID: apple:1006:10-12                                          -86dBm 491de7904368
1 New   | MAC: ce373432365d, ID: name:gvh5101_365d                                         -73dBm 
1 New   | MAC: 50488576d8ba, ID: apple:1006:10-12                                          -95dBm 181e5199fba4
1 New   | MAC: d33932353028, ID: name:gvh5101_3028                                         -79dBm 
1 New   | MAC: d83932356966, ID: name:gvh5101_6966                                         -71dBm 
Disconnected from MQTT; reason 0
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
1 New   | MAC: 6c81db26f266, ID: 6c81db26f266                                              -70dBm 
1 New   | MAC: 6f205039719f, ID: 6f205039719f                                              -62dBm 
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
1 New   | MAC: 593c8475a8b0, ID: apple:1006:10-12                                          -92dBm 141e2fdc895e
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
1 New   | MAC: 11ce7ec1a430, ID: msft:cdp:0902                                             -80dBm 
Low memory: 19528 bytes free
1 New   | MAC: 53dd2b827809, ID: apple:1006:10-12                                          -86dBm 461d5994a408

abort() was called at PC 0x401ba036 on core 1

Backtrace:0x40083d19:0x3ffe56100x40095fed:0x3ffe5630 0x4009b389:0x3ffe5650 0x401ba036:0x3ffe56d0 0x401b7aef:0x3ffe56f0 0x4019f856:0x3ffe5710 0x400fde0c:0x3ffe5730 0x400fc952:0x3ffe5770 0x400dee66:0x3ffe57a0 0x400def34:0x3ffe5800 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

nalkinburgh, 2022-08-31:
I updated the device to 3.0.2 and continue to have the same problem.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Firmware:     esp32
Version:      v3.0.2
WiFi Scan:    All (picked channel=7 bss=28:EE:52:A7:C2:23 rssi=-69)
IP address:   10.0.0.234
DNS address:  10.0.0.1
Hostname:     espresense-kitchen
Room:         Kitchen
Mqtt server:  10.0.0.233:1883
Max Distance: 16.00
Init Free Mem:139128
PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    

1 Motion| Pir: no Radar: no
1 New   | MAC: a4c138d1e55d, ID: a4c138d1e55d                                              -82dBm 
1 New   | MAC: d33932352d41, ID: name:gvh5101_2d41                                         -75dBm 
1 New   | MAC: a4c138ddb14c, ID: a4c138ddb14c                                              -93dBm 
1 New   | MAC: 5bfe9c892fa8, ID: apple:1005:9-24                                           -89dBm 01183c9d69
1 New   | MAC: 73a7e29f6a7e, ID: apple:1006:10-12                                          -75dBm 171ed0a90eb8
1 New   | MAC: 67cb54357176, ID: apple:1005:9-12                                           -73dBm 0f1c0c2bbf
1 Close | MAC: 54b25c47c2fc, ID: apple:1007:11-8                                           -36dBm
1 New   | MAC: 54b25c47c2fc, ID: apple:1007:11-8                                           -36dBm 201fdf27db3a58
1 New   | MAC: 52e6fe5fe6ce, ID: apple:1005:9-24                                           -52dBm 02984c1252
1 New   | MAC: f80dac927b75, ID: f80dac927b75                                              -73dBm 
1 New   | MAC: 38f13b40af41, ID: msft:cdp:0902                                             -84dBm 
1 New   | MAC: d83932356966, ID: name:gvh5101_6966                                         -68dBm 
1 New   | MAC: ce373432365d, ID: name:gvh5101_365d                                         -76dBm 
1 New   | MAC: d93932353e41, ID: name:gvh5101_3e41                                         -70dBm 
1 New   | MAC: 734388b6832c, ID: apple:1006:10-12                                          -88dBm 481d12ddd138
Disconnected from MQTT; reason 0
1 New   | MAC: 4d06c0b3684e, ID: apple:1006:10-12                                          -91dBm 101e92687a82
1 New   | MAC: 08115bcdf0dc, ID: msft:cdp:0902                                             -88dBm 
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
1 New   | MAC: d33932353028, ID: name:gvh5101_3028                                         -78dBm 
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
1 Left  | MAC: 54b25c47c2fc, ID: apple:1007:11-8                                           -88dBm
0 Reconnect timer
0 Reconnecting to MQTT...
1 Close | MAC: 54b25c47c2fc, ID: apple:1007:11-8                                           -34dBm
Disconnected from MQTT; reason 0
0 Reconnect timer
[... 180 lines cut ...]
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Firmware:     esp32
Version:      v3.0.2
WiFi Scan:    All (picked channel=7 bss=28:EE:52:A7:C2:23 rssi=-63)
IP address:   10.0.0.234
DNS address:  10.0.0.1
Hostname:     espresense-kitchen
Room:         Kitchen
Mqtt server:  10.0.0.233:1883
Max Distance: 16.00
Init Free Mem:139120
PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:

fila612, 2022-08-31:
same here, I came from 3.0.0b10 to 3.0.2 and the Mqtt has this problems. also the web-UI is laggy and have long breaks between the clicks.

nalkinburgh, 2022-08-31:
Interestingly I did update one node to 3.0.2 Verbose to see if I got any more information from the logs, and that one has stayed connected. My DHT11 gives no reading every other cycle, but it has stayed connected consistently for over 12 hours. None of the others will stay connected and I usually can not even access them through the IP address. It either times out, or only partially loads the settings page.

DTTerastar, 2022-09-01:
I have seen where a fresh serial install makes everything better.  Can someone else who's having this issue please try that?

nalkinburgh, 2022-09-01:
I did do that. I serial flashed 3.0.2 on the Kitchen device. That is the last logs I posted. I also did a serial flash with 3.0.2-verbose and oddly that one works.

DTTerastar, 2022-09-01:
Can you enable the status led w/ a pwn led so you can watch it flash?  I think something is getting so stuck that it's failing the mqtt timeout.

nalkinburgh, 2022-09-01:
I have one, that was not serial flashed just updated, that is status flashing. It will not stay connected. It just keeps trying then connects for less than a minute, then it is offline for anywhere from 5 minutes to over an hour. The entire time the led is flashing. I had to unplug that one because it is in my bedroom and looked like a police car had parked in there. I can't even log into them via the web browser to turn on or off the led.

DTTerastar, 2022-09-01:
Are you in a high density ble location? Is the ble so busy other stuff can't happen?

nalkinburgh, 2022-09-01:
You can pretty much see all of the bluetooth activity in the logs. I am not sure what you would consider high density. 4 phones, two watches, some bluetooth speakers and headphones in the house. It looks like I'm getting a few things from the neighbors, but nothing overwhelming and currently the only active sensor in HA is for my phone and watch.

billimek, 2022-09-01:
This is probably not related, but I'm going to throw it out there just in case:

I'm running 6 different esp32 devices (all the same) and every single one of them were not able to retain a stable connection to the (ubuiquiti) AP if that station was using the 2.4GHz channel 6.  I even tried a completely different AP and as soon as I switched it to channel 6, all the ESP32s had tremendous problems staying connected.  No issues whatsoever when connecting to a WiFI AP broadcasting on channel 1 or channel 11.

In the screenshot below, we see the top ping result of an ESP32 connected to an AP on channel 6 and the bottom on channel 11.  I can very consistently reproduce this condition.

[image]

DTTerastar, 2022-09-01:
Yikes!  That'll do it.  Never even knew pings times so high were possible.

nalkinburgh, 2022-09-01:
I can check if that is happening on mine, but mine are currently using channel 7 as indicated in the logs.

DTTerastar, 2022-09-01:
Use 1,6 or 11, never anything else.  That's 2.4G WiFi gospel.

AgentHawk, 2022-09-02:
Is Power saving activated in ESPresense?

DTTerastar, 2022-09-02:
Yes we have to use modem sleep on WiFi in order for antenna sharing w/ BLE to work properly.

nalkinburgh, 2022-09-02:
So I have forced the WiFi channel to 1 and they all connected back up and stayed connected. I have no idea why this update kicked them all off or why they wouldn't connect on that channel, but it at least is working again. Not sure if you consider this closed with that or not.

DTTerastar, 2022-09-02:
Maybe someone with some fancy WiFI sniffing equipment will help figure it out

AgentHawk, 2022-09-02:
It works on my side with Unifi and Channel 6 after a fresh install. Erase first.

DTTerastar, 2022-09-02:
Starting to think this is related: https://github.com/ESPresense/ESPresense/issues/503

AgentHawk, 2022-09-02:
try to test this

AgentHawk, 2022-09-02:
@nalkinburgh are you using the newst version? 3.0.4?
can't reproduce it on the current version

nalkinburgh, 2022-09-02:
I had this issue with 3.0.0 - 3.0.2. I am still on 3.0.2 for most of mine but do have one still on auto update that is on 3.0.4 but has not been tested on anything but channel 1.

AgentHawk, 2022-09-02:
yeah....had the same issue with 3.0-3.0.2.

Try the latest - I think its gone

billimek, 2022-09-02:
Confirming that this seems to be working as expected now as of v3.0.4 when an AP using the wifi 2.4GHz channel 6

DTTerastar, 2022-09-02:
I put in a fix for HT_40 being used for the soft_ap: https://github.com/ESPresense/AsyncWiFiSettings/commit/966c970a8eb69b17905c120e05406d35a3bbd248

DTTerastar, 2022-09-05:
Alpha now has this build to try.  The earlier code didn't compile...

DTTerastar, 2022-09-05:
https://github.com/espressif/arduino-esp32/issues/6654 this seems to track w/ your experiences, might want to pipe up in there.  Arduino 2 is new in [v3.0.0b3](https://github.com/ESPresense/ESPresense/releases/tag/v3.0.0b3)

DTTerastar, 2022-09-07:
I worked a bunch on WiFI reliability in the betas and I am seeing better performance from the 2 devices that had the weakest WiFi signal.

closed 2022-11-14 by stale[bot]

hickmanz, 2022-12-21:
On 3.1.1 I experience the following:

Channel 6: No connection to AP even at 2 feet
Channel 11: unifi saying 60% connection quality at 2 feet from AP (15 feet no connection)
Channel 1: Unifi says 100% connection quality at 2 feet and 15 feet through walls

hickmanz, 2022-12-21:
Still VERY long ping times to both devices regardless of proximity (80-800ms)

commit d9d43815 (2022-09-05): Attempt to fix #628
commit 73d8c973 (2022-09-06): Attempt to fix #628