ESPresense/ESPresense #630: Status - blue - led no longer blinking after update to 3.0

onlyoneme, 2022-08-30:
I cannot get my status led blinking the way it was working on 2.x. I use ordinary dev esp32 boards with blue led on gpio 2. All I can do is static flash or no led at all with -1 set on relevant pin.

DTTerastar, 2022-08-30:
Finally someone who appreciates a bit of status blinking!  

Can you try installing the pwm branch: https://espresense.com/alpha

DTTerastar, 2022-08-30:
I've tested it on 2 of mine and i'm convinced enough to release it.

onlyoneme, 2022-08-30:
It's blinking now, thx. But flash power seems to be much lower compared to constant flashing on inverted setting. Is it by purpose?

DTTerastar, 2022-09-01:
Yeah, it's PWM now so we can have more than on or off.  I default it to lower brightness since loads of people complain about the brightness.

onlyoneme, 2022-09-02:
> Yeah, it's PWM now so we can have more than on or off. I default it to lower brightness since loads of people complain about the brightness.

Would you consider adding a parameter to control it? My led is barely visible now tbh.

DTTerastar, 2022-09-02:
You can change it via MQTT already, but it needs to save/restore on boot or else you'll have to keep setting it.

onlyoneme, 2022-09-02:
> You can change it via MQTT already, but it needs to save/restore on boot or else you'll have to keep setting it.

It looks "inverted" there. I have just 1 led enabled but in MQTT I can see:

[image]

DTTerastar, 2022-09-05:
There is an option for PWM inverted, select that.

DTTerastar, 2022-09-05:
#654 fixes the extra LEDs in mqtt

onlyoneme, 2022-09-06:
> #654 fixes the extra LEDs in mqtt

Wiil it also fix incorrect message for active led state, shown as "OFF"?

DTTerastar, 2022-09-06:
No, I think you need to switch it in here:
[image]

onlyoneme, 2022-09-06:
> No, I think you need to switch it in here: [image]

When I switch to inverted it stops blinking...

DTTerastar, 2022-09-06:
I'm stuck.  I've tested my 2 PWM led boards (one is inverted the other isn't) and both blink the status like when the LED control is set to "Status".

onlyoneme, 2022-09-06:
> I'm stuck. I've tested my 2 PWM led boards (one is inverted the other isn't) and both blink the status like when the LED control is set to "Status".

I don't know what to say... When my led is enabled as normal PWM it blinks with low brightness  and it sends status = OFF after the restart. When I set the led as inverted PWM it stops blinking after the restart, has full brightness and also sends status = OFF.
I have an ordinary esp32 dev board identified as ESP32-D0WDQ6 with led on default pin 2.

onlyoneme, 2022-09-06:
I've upgraded to 3.0.5b2. Nothing has changed, still incorrect messages around leds.

DTTerastar, 2022-09-07:
state = OFF is irrelevant for status control, we can't send ON/OFF every 0ms when we are flashing on and off.

onlyoneme, 2022-09-07:
I thought that state reflected enable status not flashing status, now I understand. But back to the point, I cannot change the brightness over mqtt. I try to send a message on led_1 topic but there is no visible effect of any changes.

onlyoneme, 2022-09-07:
I think it's working on 3.0.5b3.
Is it correct to send message like below to /set topic for selected led?
{
  "brightness": 255
}

closed 2022-11-14 by stale[bot]