ESPresense/ESPresense #634: Question regarding irk feature

onlyoneme, 2022-08-30:
Is it supposed to work also with android devices?

51av0sh, 2022-08-31:
Yes it works with Android but I've noticed it takes longer (in my case) to detect presence vs using iBeacon*. I have two android phones and tested moving to different rooms with both on my hand. The one using iBeacon was changing rooms faster every time. In average it was taking 5 second longer with the phone using the irk number

onlyoneme, 2022-09-01:
Honestly it doesn't work for me with android. I can pair my phone after enroll on espresense, I can even see irk generated in settings, but that's all. Phone is not being tracked.

janssenjones, 2022-09-01:
> Honestly it doesn't work for me with android. I can pair my phone after enroll on espresense, I can even see irk generated in settings, but that's all. Phone is not being tracked.

You may already know this, but just in case - The mechanism actually changes when you use the enroll method (this also changed a bit in recent versions) - it actually starts sending the enrolled friendly device name to MQTT rather than the other names - so for example if you enrolled and called it myPixel, then myPixel is what would start getting sent to MQTT, and also what would be showing up in the espresense/ui URL rather than the IRK itself.  Though I've not personally tested on Android, as I only have iPhones around.  But just wanted to make sure you aware in case that was causing any confusion. So you'd need to change your sensor in Home Assistant to be that friendly name as well.

onlyoneme, 2022-09-01:
@janssenjones 
Yes, I've seen that enroll from /ui subpage works a bit different that enroll button in HA. But my android phone is not being detected regardless of enroll method.
With iPad it works as expected but not with android. At least not for me.

51av0sh, 2022-09-01:
So I didn't use the /ui subpage. I also couldn't get the irk number to work initially. It took several reboots, adding, removing, updating devices and something made it work but I couldn't tell you what. There was nothing in the logs, the irk was showing up in mqtt with no issues. I removed them and only using irk with my wife's iphone now though.

DTTerastar, 2022-09-01:
Irks will work w/ android as long as the android has a reason to advertise.  So you STILL need an app most of the time, which kinda defeats the purpose in my opinion.

onlyoneme, 2022-09-01:
@DTTerastar 
Thx for the explanation.

closed 2022-09-01 by onlyoneme