ESPresense/ESPresense #635: No AP appears following firmware upload on ESP32-C3-12F-Kit

shane64, 2022-08-31:
No access point is created after installing from source (vscode/platformIO).  SSID and IP Address are available on the serial console but no device (phone nor laptop) can see the SSID over the air.

**Config**

- Version: 3.0.2 uploaded via vscode/platformIO

**Hardware**

- Device (be specific): ESP32-C3-12F-Kit (4MB version)

**Logs**

Debug level 4 - console output
```
23:46:55.619 > ESP-ROM:esp32c3-api1-20210207
23:46:55.619 > Build:Feb  7 2021
23:46:55.619 > rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
23:46:55.619 > SPIWP:0xee
23:46:55.619 > mode:DIO, clock div:1
23:46:55.619 > load:0x3fcd6100,len:0xf4
23:46:55.619 > load:0x403ce000,len:0x8a0
23:46:55.619 > => 0x403ce000: ?? ??:0
23:46:55.619 > load:0x403d0000,len:0x218c
23:46:55.619 > => 0x403d0000: ?? ??:0
23:46:55.619 > SHA-256 comparison failed:
23:46:55.619 > Calculated: 402ae7f8d96ee2a56a5a78dd380569c46de7b91df4dfdb33de30b80f17413178
23:46:55.619 > Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
23:46:55.619 > Attempting to boot anyway...
23:46:55.619 > entry 0x403ce000
23:46:55.619 > => 0x403ce000: ?? ??:0
23:46:55.619 > Setup network
23:46:55.619 > [   234][E][esp32-hal-i2c.c:87] i2cInit(): i2c_param_config failed
23:46:55.619 > [   286][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
23:46:55.619 > [   318][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
23:46:55.619 > First contact!
23:46:55.619 > 
23:46:55.619 > [   325][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
23:46:55.619 > [   329][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
23:46:55.619 > [   336][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
23:46:55.619 > Starting access point for configuration portal.
23:46:55.619 > SSID: 'espresense-0c953a'
23:46:55.620 > [   348][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
23:46:55.620 > [   350][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
23:46:55.620 > IP: 192.168.4.1

```
To confirm that the hardware works, I loaded a Tasmota 12.1.1 binary via the ESP-Flasher tool and was able to see the AP, connect via config portal (to 192.168.4.1), configure the local SSID to connect to, reboot and it connected to my network AP fine (whereby I could proceed to configure further options).  I am unable to do so with ESPresence 3.0.2 - no SSID shows up over the air.

Thank you in advance.

DTTerastar, 2022-09-01:
Can you see if tasmota is setting up anything special for wifi for the c3?  You're not crazy some C3 work and some don't.  I have both but not sure what the difference is.

DTTerastar, 2022-09-07:
I experienced this today and I tried all sorts of things to fix it.  Erasing the whole flash and then reprogramming and it worked just fine again.  Super weird.

closed 2022-11-14 by stale[bot]