ESPresense/ESPresense #640: Apple Watch recognition stuck

AgentHawk, 2022-09-01:
Hey

noticed the behavior in the beta aswell.

It works if you start the ESP for a couple of Minutes. Sometimes more. And if you leavin or turning the BT off it like he ignores the Watch completly. Nothing via WebTerminal same with mqtt.
Not a single ping from the Watch.

But if you restart espresence everything works as usual.

3.0.3
ESP32 WROOM-32

AgentHawk, 2022-09-01:
Reproducible

Start ESPresence - Turn Off BT on your Watch - turn it back on - Blacklisted - restart fixed the issue

DTTerastar, 2022-09-01:
How long do you need to turn off BT on the watch?

AgentHawk, 2022-09-01:
IRK via Apple Keychain.

BT is off and on (maybe 1 or 2 sec)

DTTerastar, 2022-09-01:
I found the issue. Please try updating to the alpha via http://espresense.com/alpha

AgentHawk, 2022-09-01:
Yep! Nice Work.

closed 2022-09-05 by DTTerastar