ESPresense/ESPresense #646: cant switch off LED after upgrading to v3.0.4

bicmac, 2022-09-02:
last night i found that all my espresence devices blinking. I checked today and found that they was upgraded to Version 3.04. 
Before in vrssion 2.9.X my LED on die ESP32 was disabled and not blinking or whatever. Now i cant switch off the LED anymore (i setuped -1 for LED as it is descibed but its still on. With that i cant use it right now as my wife and mid kid comlain abozt the light. I will coverLED now for the moment with back tape but woulf be great if we can somehow find a way to dayable that LED again in Admin page.

onlyoneme, 2022-09-02:
Did you try to power off and on your devices after disabling leds in setup? Restarting them is not enough.

bicmac, 2022-09-02:
> Did you try to power off and on your devices after disabling leds in setup? Restarting them is not enough.

yes i tried, but it is is still on.

DTTerastar, 2022-09-05:
If it's PWM and inverted you may need to set the led pin but put the control as mqtt

closed 2022-11-11 by stale[bot]

ChrWieg, 2024-10-13:
Didn't work with my esps. I set all three LED pins to -1. I think the red LEDs on my devices are other LEDs that may indicate the supply voltage.

ku1ik, 2025-04-14:
Many esp32 boards have 2 built-in LEDs: red and blue. The red is often related to the USB power, and can't be turned off, while the blue one is often connected to PIN 2 and can be controlled via firmware.