ESPresense/ESPresense #647: Add password protection to admin page of ESP Device

bicmac, 2022-09-02:
**Is your feature request related to a problem? Please describe.**
unprotected Admin Page of ESP32 device is a security Risk.

**Describe the solution you'd like**
Setup a user and Password protection for the ESP32 Devices.
in best case with2 Roles:

Admin who can see and Edit config
GUEST: who can just see actual config

**Describe alternatives you've considered**
i checked to separate the network but this is not really possible in my actual setup.

**Additional context**
Add any other context or screenshots about the feature request here.

hawithme, 2023-05-13:
Critical feature

codyc1515, 2023-05-13:
Let's say they added password protection to the UI. What about remote flashing? There's myriad other ways around this. Ultimately, this software is really only intended for use on a secure network.

peter-dolkens, 2023-05-14:
> Let's say they added password protection to the UI. What about remote flashing? There's myriad other ways around this. Ultimately, this software is really only intended for use on a secure network.

You're right - and we should just get rid of keys from houses and cars all together - they don't do anything against a real criminal with more tools than your average junkie on the street.

/s

ndbroadbent, 2023-05-15:
This would be nice to have. I guess it's not too important but I don't like having anything unsecured where visitors can break things, either on purpose or by accident. Especially if we people staying at our house or housesitting for us.

creuzerm, 2023-05-27:
I just pushed new firmware onto my units today, the latest beta, and we can just tell it 'use this firmware from an arbitrary url' so it is trivial to compromise these units.

REALLY HANDY to be able to upgrade to your own custom code without needing to physically dig the units out of whatever hidey hole they are tucked away in. But really scary that somebody sitting on the street can crack the wifi password and put anything they want on my ESP32s.

I like how ESPHome uses the keys and such. I've not validated the security on this method but I expect even if it's not perfect, it's not trivial either. We don't currently have a good place to put those keys. I see commentary around ESPresense Companion that a goal is to move towards node configuration (If my memory serves correctly).

closed 2023-09-19 by stale[bot]

gith-account, 2024-07-08:
> Ultimately, this software is really only intended for use on a secure network.

If the device loses the wireless network connection, either due to a technical fault or actively triggered by a third party without physical access to the device or access to the wireless network, it makes itself openly available to the public by means of its own wireless AP, outside of the secure network which it may later re-enter in a compromised state. This is a significant security vulnerability that can only be prevented by disabling the capture portal in the event of network loss, which significantly impairs maintainability. The password protection proposed here would improve security considerably.

gith-account, 2024-07-08:
Please reopen the issue.

gith-account, 2024-07-18:
@ESPresense, @DTTerastar: May I kindly request that you reopen this issue?