ESPresense/ESPresense #650: Issues configuring Apple Watch Detection

jmcgill, 2022-09-03:
I've gone through the instructions to register the IRK for my Apple Watch, but have not got it to successfully register. My goal is to identify whether there's an improvement that can be made to the documentation, whether there's an edge case I can help to debug, or whether I'm just bad at following instructions!

I am very excited and impressed by the IRK functionality and eager to dive into it!

I am knowingly pasting my own IRK details here - if someone wants to follow me around, more power to you! 

To try and remove as many variables as possible, I am looking at the MQQT topics directly using MQQT explorer.

**Steps taken**
Extracted the XML configuration from iCloud Keychain, confirming that the MAC address shown in the "Account" field in Keychain matches the mac address shown in Settings on my iWatch.

[image]

The result is the following XML document:

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Remote IRK</key>
	<data>
	WgHV+w44IiVKv3ca4ug1yQ==
	</data>
</dict>
</plist>
```

Decoding the base64 for this IRK (using https://cryptii.com/pipes/base64-to-hex) gives:

5a 01 d5 fb 0e 38 22 25 4a bf 77 1a e2 e8 35 c9

Reversing the endianness gives:

c9 35 e8 e2 1a 77 bf 4a 25 22 38 0e fb d5 01 5a

For a final IRK string of:

irk:c935e8e21a77bf4a2522380efbd5015a

For good measure, I repeated the process in python to try and eliminate user error.

```
import base64
h = base64.b64decode('WgHV+w44IiVKv3ca4ug1yQ==').hex()
r = [h[i:i+2] for i in range(0, len(h), 2)]
r.reverse()
print('irk:' + ''.join(r))
```
I added this value to the "Known BLE identity resolving keys" section in the configuration UI and saved then rebooted the device for good measure:

[image]

At this point, I expected a new device to show up in MQTT explorer under the espresence/devices topic with a reference to the provided irk, however nothing shows up.

[image]

Same story on the fingerprints UI

[image]

**What I've tried**
I've taken a look through a host of related issues and tried what I can:

https://github.com/ESPresense/ESPresense/issues/460#issuecomment-1220287957
https://github.com/ESPresense/ESPresense/issues/649
https://github.com/ESPresense/ESPresense/issues/640
https://github.com/ESPresense/ESPresense/issues/33

I also tried skipping the reversing step to see if that helps. No dice.

**Config**

- Version: I've tried the latest release, and the alpha (currently 622bf52)
- Active scan enabled: No
- Include filter: N/A
- Exclude filter: N/A
- Query filter: N/A

**Hardware**
 
- Device (be specific): D1 Mini NodeMCU ESP32

**Logs**

I didn't see anything relevant to IRK, even with the Verbose image - however I might be missing something and can export if useful.

nalkinburgh, 2022-09-03:
I think...I could be wrong...that in the known BLE field you should only put the hex pairs and not include "irk:" That should be included in the sensor portion of your YAML config.

jmcgill, 2022-09-05:
Thank you, that worked! I've submitted a PR to update the documentation.

closed 2022-09-05 by jmcgill

DTTerastar, 2022-09-05:
Multiple irks should be space seperated, not comma

Buckeyes1995, 2022-09-05:
Space separated did not work for me I’m the 2. Betas.  Had to put a comma in.  Maybe they changed?

rhamblen, 2023-06-30:
is there anyway to get the IRK without an apple mac?  
can it be triggered a different way similar to the way to enroll an iphone or ipad?

Aviv943, 2023-07-15:
> is there anyway to get the IRK without an apple mac? can it be triggered a different way similar to the way to enroll an iphone or ipad?

Have you found an alternative way to obtain the Apple IRK without using a Mac?

DTTerastar, 2023-07-16:
I haven't really tried since I've got lots of macs.  Is there a way of accessing iCloud keychain on a PC?

Aviv943, 2023-07-19:
Unfortunately not, iCloud Passwords only. there isn't an option to search
for Bluetooth devices

DTTerastar, 2023-08-31:
Go to an apple store and play with a mac for a bit maybe?