ESPresense/ESPresense #651: Support automatic decoding of Base64 strings for Apple Watch

jmcgill, 2022-09-03:
**Is your feature request related to a problem? Please describe.**
There are several steps required today to convert a Base64 encoded IRK into an IRK useable by ESPresence. These steps might be particularly difficult for folks unfamiliar with hex encoding / endianness. 

**Describe the solution you'd like**
1. Detect when an IRK is provided as a base64 string in configuration and automatically decode it. 
2. Display the decoded IRK string in the devices UI so that it can be copied into HASS

**Describe alternatives you've considered**
* Add a separate input field for Apple Watch
* Build a decoding utility into the /ui portion of the app
* Create a new prefix (e.g. b64irk:) for base64 encoded strings

**Additional context**
I would gladly implement this, and have a draft change ready to go, but need to nail down the solution to https://github.com/ESPresense/ESPresense/issues/650 first so I can test it end to end.

DTTerastar, 2022-09-05:
The flow I would like is to get rid of known mac and known irk, and add a button next to Enroll that is "Add IRK" and "Add MAC".  On that box we could accept both the raw bytes or the base64 format and normalize them.  Svelte is the gui used in /ui/

DTTerastar, 2022-09-05:
I also want to support adding a name

antongiacomo, 2022-09-05:
I can start working on something concerning the ui on a separate branch, if that's okay with you.

DTTerastar, 2022-09-05:
Sure

closed 2022-11-11 by stale[bot]