ESPresense/ESPresense #652: Adding support for the ESPHome Bluetooth Proxies

madrose, 2022-09-04:
I have started using ESPresense recently and it is working like a charm to track my Mi Band, and I also started using the new Bluetooth feature in HA, and this required me to have one ESP32 for the Bluetooth proxy and one for the ESPresense. 

Why not having the ability to add the Bluetooth proxy and ESPresense in one ESP32 device rather than having to use 2?

DTTerastar, 2022-09-05:
Yeah I was looking into that myself.  We'd need to make an mqtt->ble event adapter or wait for an official one.  ATM only noise is supported I think.

javellino, 2022-09-05:
Is it possible to use espresense as a custom_component on esphome platform? 

https://esphome.io/components/sensor/custom.html

DTTerastar, 2022-09-05:
#233 No...

ndbroadbent, 2022-09-22:
I recently set up 5x ESP32s around my house to run ESPresence, and I'd really like to use them to get data from my passive BLE temp/humidity sensors ([Xiaomi lywsd03mmc](https://github.com/ESPresense/ESPresense/issues/508)). It would be great if we could use these as a bluetooth proxy with Home Assistant, or even if it could just scan for these devices and create it's own temp/humidity sensors in MQTT.

I'm really impressed with the Apple device support in ESPresence and the overall experience, and I've found it impossible to reimplement these features in ESPHome. But it would be awesome if the projects could work together in some way so that we can use features from both

MichaelStruck, 2022-10-10:
I agree...ESPresense AND BLE Proxy would be great. Otherwise, we would have to double the number of ESP32s in the house. I envision the MQTT page for the device simply having a check box to also act as a BLE proxy in HA. Of course, it is never as simple as what I envision it. Have to wait for @DTTerastar to chime in if it 100% impossible or just would take some time to implement.

NicholasMantovani, 2022-11-28:
> I have started using ESPresense recently and it is working like a charm to track my Mi Band, and I also started using the new Bluetooth feature in HA, and this required me to have one ESP32 for the Bluetooth proxy and one for the ESPresense.
> 
> Why not having the ability to add the Bluetooth proxy and ESPresense in one ESP32 device rather than having to use 2? Can ESPresense be installed as package in ESPHome?

Sorry for asking which mi band are you using? I want to buy the 7 do you know if that work?

madrose, 2022-11-28:
> Sorry for asking which mi band are you using? I want to buy the 7 do you know if that work?

I am using the Mi Band 6, but I believe Mi Band 7 will work as well.

SuperDobbo, 2023-01-14:
Another vote for adding BlueTooth proxy, so far I have 8 presence devices with 4 more to add, I really dont want to double up devices.

Shesakillatwo, 2023-01-22:
Just curious if this has in anway moved forward or if it will.  I have 5 Bluetooth proxies in my house and would love to load ESP Presence on these rather than add additional Bluetooth devices just for the presence capability.  Thanks!

mlava, 2023-03-01:
+1 here as well.
Is there anything we can do help this along?

luke8804, 2023-03-12:
+1 to this request, I’d love to see them integrated!

mlava, 2023-03-13:
@DTTerastar 
Is there any way we can encourage this idea? I'm happy to donate $
😃

aviadlevy, 2023-03-27:
I fully support this feature request. As a user of this project, I would find this to be incredibly useful for many users.  
I hope that the maintainers can consider implementing this feature in a future release. Thank you for considering this request.

peewi96, 2023-04-01:
+1 this would be awesome and would allow me to put more ESPresense devices around the house 👍

CumpsD, 2023-06-26:
Same here, I got a bunch of olimex boards around the house, setup with ESPHome as a Bluetooth Proxy for Home Assistant. Recently started looking into room-level presence detection and found ESPresense and Room Assistant. But neither seems to be able to make use of the same devices already installed. I'm powering them with PoE, and not really eager to duplicate everything.

Can we debug anything to somehow make this happen?

DTTerastar, 2023-06-27:
Some investigation on what sort of data format hass accepts passive ble data in as.  If it's noise only, it's a none starter.  I've only seen official integrations use the ble proxy feature so I don't know if it's even available for us to use at all.

CumpsD, 2023-06-27:
> Some investigation on what sort of data format hass accepts passive ble data in as. If it's noise only, it's a none starter. I've only seen official integrations use the ble proxy feature so I don't know if it's even available for us to use at all.

I'm wondering if something can be done with the `on_ble_advertise` hook from https://esphome.io/components/esp32_ble_tracker.html where it perhaps can send the data to ESPresense to deal with it?

CumpsD, 2023-06-27:
> That uses noise to talk to home assistant. It'd be easier to just flush the current ESPresense and make it a component of ESPHome.

Indeed, I'm now looking into https://github.com/esphome/esphome/tree/dev/esphome/components/bluetooth_proxy to see what they are actually sending to HA. If ESPresense can send the same thing, HA will think it's just a regular bluetooth proxy.

CumpsD, 2023-06-27:
I managed to trace it to this point:

* https://github.com/esphome/esphome/blob/dev/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp#L34C20-L34C20
* https://github.com/esphome/esphome/blob/dev/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp#L84
* https://github.com/esphome/esphome/blob/dev/esphome/components/api/api_connection.cpp#L875
* but then I got lost at: https://github.com/esphome/esphome/blob/d9398a91d1dac8e95e7a7e135a24be50e124bc5b/esphome/components/api/api_pb2_service.cpp#L338

DTTerastar, 2023-06-28:
You missing that everything in ESPHome uses noise to talk to home assistant.  Noise is not an option for ESPresense since it's all based on MQTT.

CumpsD, 2023-06-28:
> You missing that everything in ESPHome uses noise to talk to home assistant. Noise is not an option for ESPresense since it's all based on MQTT.

I indeed didn't catch that :(

closed 2023-10-11 by stale[bot]

PCServices2, 2024-01-02:
I too am looking for either a way to add Bluetooth Proxy to the ESPresense ESP32's or add ESPresense function to my Bluetooth Proxy ESP32's.
The second option would be best so as to take advantage of the functions on the RSPHome integration.
I had hoped to find some code to add to the configuration for the Bluetooth Proxy devices.

ddhoch, 2024-02-17:
Still looking for a solution to combine both of these functions in one device. Bluetooth Proxy and ESPresense. I  too am willing to throw a few bucks in the right direction to get this started.  Let's get this moving.

tempo3, 2024-05-25:
+1 this would be great not to double the ESPs

constapel, 2024-06-22:
+1 please consider prio for this project: ESPresense and Bluetooth proxy. Use case is: personilaztion in mmwave ble sensors and ble beacons. E.g. Smart Lighting only when Smart Ring or Smart Watch is in neigbourhood....

geoff035, 2024-11-18:
I'm not very clued on as to how all the ble stuff works but my understand is that the bt proxy just sends to HA what it receives from the ble device. Could it not just be sending the same info the espresence receives (I'm guessing just the mac address and rssi) from the beacon and send it to HA to do the heavy lifting? Or is this pretty much what the Bluetooth LE tracker in HA already does?

desmond206x, 2025-05-15:
I thought ESPresence is part of ESPHome.. So please, make it possible to use ESPresence inside of ESPHome