ESPresense/ESPresense #657: Apple Device Enrolled via IRK showing as 'Not_Home'

leacho73, 2022-09-06:
I've just upgraded to the latest ESPresense Beta and I have enrolled my iPhone using the instructions here: https://espresense.com/beacons/apple - however, even though I can see my IRK in MQTT Explorer - both under:

Rooms -> Kitchen -> Settings -> IRK
Rooms -> Kitchen  -> known_irks

When I add the IRK to my Home Assistant Config:

- platform: mqtt_room
  name: "My iPhone"
  device_id: "irk:###################[Redacted]"
  state_topic: "espresense/rooms"
  timeout: 5
  away_timeout: 120

I still get 'not_home' shown in Developer Tools - its also worth noting that although I can see the IRK and phone inside MQTT Explorer - its not listed under Home Assistant -> Integrations -> Devices.

Any ideas?

Cheers

javellino, 2022-09-06:
If you look under /ui and click on fingerprints you should see the alias you used in the list. I use the alias in homeassistant and it works for me.

javellino, 2022-09-06:
> Did you include the IRK in your Known BLE devices in the ESPresence?

Negative. For an iPhone you just have to enroll it at one of the espresense
devices. The others pick it up by the mqtt espresense/settings topic. I
think the beacons page was updated recently.

This works for me on v 3.0.4. If you’re not on that version you might want
to consider updating

leacho73, 2022-09-07:
Hi Both,

I can see the Alias on the Fingerprints page of the ESP - I've used the Alias as per the code snippet above - and I can see my sensor.my_iphone in home assistant - but it just show's up as 'not_home' - even though I can see on the fingerprints page that the distance for the phone is changing, so the ESP appears to recognise that its there.

javellino, 2022-09-07:
https://espresense.com/beacons/apple

I assume you do not have the IRK in the 'known BLE revolving keys' field
correct?  this documentation was updated.

Also what version are you on?

can you post your mqtt_room sensor code snippet?

Also have you tested using a different BLE token you may have around your
location?

leacho73, 2022-09-07:
Hi @javellino 

Version is v3.0.5b1 - my IRK is in the list of known BLE revolving keys - and I can confirm this using MQTT explorer - code snippet used in my sensors.yaml file is as follows:

platform: mqtt_room
name: "My iPhone"
device_id: "irk:###################[Redacted]"
state_topic: "espresense/rooms"
timeout: 5
away_timeout: 120

I will try it with another Apple device in a moment.

Cheers
Leacho

javellino, 2022-09-07:
instead of using the irk as the device_id, try using the alias you used.
for example 'joesiphone' instead of 'Irk:xxxxxxx...xxxx'

Also per the documentation at https://espresense.com/beacons/apple please
note

Note: With the automatic procedure you DON’T need to add the 32-bit key to
the Known BLE identity resolving keys. It’s automatically done via mqtt!

you can take that out.  I know this works with 3.0.4.  I have not tried
with 3.0.5.b1, but this has been pretty reliable for me with multiple
iphones and multiple devices.

leacho73, 2022-09-07:
Changing the alias in the yaml sorted it - thanks @javellino - I've also removed the keys from the known BLE field and its now identifying correctly for multiple devices.

Do I have to enroll the devices on each individual ESPresence ESP for them to be recognised? I've got quite a few :)

Thanks again for all your help!
Leacho

leacho73, 2022-09-07:
@javellino - Ignore the enrolling question - just re-read your earlier comment! :)

javellino, 2022-09-07:
Glad it’s working.

And no once it’s enrolled on one device the others pull it into their
configuration by virtue of the mqtt espresense/settings topic.

closed 2022-09-07 by leacho73

Zipties, 2022-09-08:
> Glad it’s working. And no once it’s enrolled on one device the others pull it into their configuration by virtue of the mqtt espresense/settings topic.
> […](#)

Does they also apply to manually added IRKs to a single ESP? Add the irk to one and the other will get it via Mqtt?

javellino, 2022-09-08:
I really don’t know. I can only attest to my use case.

DTTerastar, 2022-09-09:
Known irks are per device.  You can use MQTT explorer to put it in settings and it'll go everywhere.  3.1 will have a gui for this if you can wait.