ESPresense/ESPresense #662: PIR pin always pulled high

Zipties, 2022-09-08:
I have a PIR sensor that while the out put is disconnected from the ESP it is functioning correctly. As in I get 3.3 volts on the out pin when there is motion. 

But when I configure the PIR in settings and connect it to a GPIO pin (I have tried 19 22 and 23) the pin is always pulled high. 

I’m fairly new to microcontrollers but this seems to indicate the pin need a pull up resistor enabled

I feel like I’m doing something wrong but I’m not sure what..

DTTerastar, 2022-09-08:
https://github.com/ESPresense/ESPresense/pull/663 has a fix.  You can install it via https://espresense.com/alpha

Zipties, 2022-09-08:
> #663 has a fix. You can install it via https://espresense.com/alpha

That did it, you're awesome

DTTerastar, 2022-09-15:
Released in v3.1.0

closed 2022-09-15 by DTTerastar