ESPresense/ESPresense #664: Samsung watch 3 not working anymore

xwox, 2022-09-08:
Hello,
I am using ESPresense since many months. But just recently I cannot see my samsung watch anymore.
It does not appear in the MQTT under devices.
I run on the latest firmware 3.04

What do I do wrong? What can I fix?

Regards,

DTTerastar, 2022-09-12:
Check http://{ip}/ui/ and see if maybe it's under another id?

xwox, 2022-09-13:
Thx for the nice address. I was not aware of this feature.

But no no watch there. Nothing that starts with "samsung:". But also no other device.

DTTerastar, 2022-09-15:
Put your watch RIGHT up to the esp32 antenna.  What id is the closest?

xwox, 2022-09-15:
Nothing really close if the watch is next to the antenna:
2,39m | exp:20 |   | 20eda2beb215 | -89dBm | -77dBm | 505ms
5,00m | apple:1006:10-7 |   | 6478712ee5db | -91dBm | -65dBm | 220ms
5,20m | d428578aafbb |   | d428578aafbb | -90dBm | -65dBm | 2769ms
7,40m | apple:1005:9-1 |   | 577dd3d35978 | -95dBm | -65dBm | 568ms
7,47m | apple:1005:9-1 |   | 5501c917c024 | -96dBm | -65dBm | 584ms
7,55m | 0c92d466f301 |   | 0c92d466f301 | -94dBm | -65dBm | 631ms
7,60m | f57707ddab7e |   | f57707ddab7e | -95dBm | -65dBm | 4047ms

closed 2022-11-21 by stale[bot]