ESPresense/ESPresense #666: Can Apple Watch IRK be added straight to MQTT with Alias?

Zipties, 2022-09-08:
Instead of adding the IRK to each ESP

[image]

Can I just do something like this and have all ESPs respect it?

[image]

DTTerastar, 2022-09-09:
Try and keep ids sluggify style (underscores for spaces all lower)

DTTerastar, 2022-09-09:
3.1 should have a ui to add these and they'll go to mqtt.

Zipties, 2022-09-12:
> Try and keep ids sluggify style (underscores for spaces all lower)

yep that was it. I was skeptical the special characters would be valid, but it seemed to work when I enrolled my iPhone, so I never tried changing it on the watch.

fila612, 2022-09-12:
@Zipties:
How do you setup your AppleWatch to get an named ID instead of the irk only?

When I configured my AppleWatch as described in the wiki ( extract the irk from Apple KeyChain), i was able put in the irk as known irk, but this irk never shows up in settings folder.

when I tried to use the enroll button (via .../ui), put a name in the field and connect via iPhone, I got the name in my config.
With an AppleWatch this doesn't work :(

choenig, 2022-09-13:
@fila612: 

For now you have to publish the setting to `espresense/settings/irk:xxxxx/config` with the payload `{"id":"my_new_id"}` by hand. 

Don't forget to set the retain flag.

AlfredJKwack, 2022-09-17:
Slightly off topic: naming of WatchOs from `http://espresense-office/ui` with [v3.1.0](https://github.com/ESPresense/ESPresense/releases/tag/v3.1.0) or better.

I've read several times that you can't pick up an IRK from the WatchOS because ESPresense can't pair with it. I don't quite get that though. The watch does have a "[Settings > Bluetooth](https://support.apple.com/en-us/HT204218)" menu. When ESPresence is in pairing mode it the watch will show the base station under the "health devices" list and successfully pair. So logically, from the `http://espresense-office/ui` gui you should be able to give it any name you wish. Right?

fila612, 2022-09-18:
> Slightly off topic: naming of WatchOs from `http://espresense-office/ui` with [v3.1.0](https://github.com/ESPresense/ESPresense/releases/tag/v3.1.0) or better.
>
> I've read several times that you [...]

No, for me that's not working. I've tried that several times - without success.

drschnalli, 2023-03-28:
how to get the IRK without a mac ?

rhamblen, 2023-06-30:
has anyone got the IRK for an applewatch without a Mac?

DTTerastar, 2023-11-04:
The native settings refuses to *securely* pair.  But using this app that someone found it works!

https://espresense.com/beacons/apple#apple-watch-using-bluetooth-terminal-app