ESPresense/ESPresense #676: DF Robot mmwave

wifispray, 2022-09-14:
Hi there.

Its not clear to me how I need to configure the web gui of the ESPresence to support the DF Robot mmwave sensor? Can someone please provide some feedback? I have created the following schematic just need some help with the config side?

[image]

DTTerastar, 2022-09-15:
IO2 goes high when detected so in your example set radar pin to 16.

Zipties, 2022-09-15:
> Hi there.
> 
> Its not clear to me how I need to configure the web gui of the ESPresence to support the DF Robot mmwave sensor? Can someone please provide some feedback? I have created the following schematic just need some help with the config side?
> 
> [image]

What are the sensors you got going on there on the right?

DTTerastar, 2022-09-15:
I think the sensor needs some kind of init before it turns on after reading the docs.  Lame.  It'll need special support.

Zipties, 2022-09-15:
Ugh, ah well back to stupid ol ESPHome for the time being :(

maromaro70, 2022-10-01:
> Ugh, ah well back to stupid ol ESPHome for the time being :(

Have you gotten anywhere with this? I could not verify if ESPresense has plans to support this seemingly accurate mmWave motions and presence sensor

addelovein, 2023-02-20:
Still nothing? Really not keen on having to run EspHome as well. 😞

addelovein, 2023-02-20:
> > I think the sensor needs some kind of init before it turns on after reading the docs. Lame. It'll need special support.
> 
> 
> 
> Any updates and/or ideas on which initial treatment is necessary? Anything we can help with?

Well the code exists in esphome repo. Or actually the custom component for it.

DTTerastar, 2026-11-04:
#2611 adds read-only UART support for DFRobot's Gravity C4001 (SEN0609/SEN0610): presence, and distance/speed/energy when the module is in speed mode. The older SEN0395 isn't covered yet; its `$JYBSS` output would be a small follow-up if anyone still uses one.