ESPresense/ESPresense #680: Are MQTT room configs still necessary?

Zipties, 2022-09-15:
I setup a new ESP node today, and my BLE endpoints all grabbed the room name configured in the ESP, but I hadn't yet configured the room in HA. 

Example of a room config I have
`mqtt:    
# MQTT Room (BLE Presence)   
   binary_sensor:
      - name: Office Room
        state_topic: espresense/rooms/office_room/status
        json_attributes_topic: espresense/rooms/office_room/telemetry
        payload_on: online
        payload_off: offline
        device_class: connectivity 
        `

DeekB1983, 2022-09-16:
According to the documentation these are no longer required.
[image]

I removed mines some time ago and can confirm the auto discovery works.

Zipties, 2022-09-16:
ah I missed that, thank you

closed 2022-09-16 by Zipties