ESPresense/ESPresense #681: how to remove IRK's??

DeekB1983, 2022-09-15:
I have been having reliability issues with Iphones since I started using ESPresense a long time ago.

I noticed yesterday when reading the documents the process seems to have changed.

I was adding a couple of devices and I seem to enrolled 2 devices using the same device name.

Is there a way to delete the IRK's after adding them so I can correct my error?

Zipties, 2022-09-15:
In MQTT delete it from espresense/settings

DeekB1983, 2022-09-16:
> In MQTT delete it from espresense/settings

I have deleted the from settings already and unpaired it within bluetooth settings of the IRK I dont want but it hasnt made any difference.
[image]

IF you look at above there is 2 Iphones paired with 2 different IRK's.

One is an Iphone 13 (IRK starting ab30 - Mirrens Iphone) 

The other is an Iphone 13 mini (IRK starting 48CA - Lauras Iphone)

Lower down in devices I also have another IRK listed starting 8f3c, This is the IRK I want to remove, you will see it does not exist within settings higher up, originally 8f3c was registered as Lauras Iphone by mistake and I am still regularly recieving messages into the MQTT Explorer from this device

Mirreens Iphone seems to be working well for the most part
[image]

But Lauras Iphone which had 2 IRK's (48ca & 8f3c) registered as Lauras Iphone seems to be on and off
[image]

You can see there is a lot of not_home, either it is not sending BLE requests\signals when the screen is off or I think it could be due to the second IRK starting 8f3c still sending messages in via MQTT.

Looking at this screenshot you can see Mirrens Iphone is receiving a lot of messages overnight but Laura Iphone is not, however the IRK starting 8f3C which was alo previously registered as Lauras Iphone is still sending lots of messages (This phone is also still in use but it has had the espresense unpaired in Bluetooth and it has been deleted from settings
[image]

Just trying to get some stability around Iphone presence but have been really struggling for a long time with these Iphones, have even changed models ect

Zipties, 2022-09-16:
Hmm, I figured that is what I had done that got rid of mine, but perhaps not. You could just delete the entire espresense root, it will just get repopulated

DeekB1983, 2022-09-17:
> 

Thanks, the IRK has not come back after a while but I seem to have the same issue I was previously which leaves me to believe that its an issue with the iphone not sending bluetooth messages when the screen is off for some reason.

Zipties, 2022-09-17:
> > 
> 
> Thanks, the IRK has not come back after a while but I seem to have the same issue I was previously which leaves me to believe that its an issue with the iphone not sending bluetooth messages when the screen is off for some reason.

I've seen my iPhone do that occasionally as well. I also seem to have over all better up time just using the generic apple identifiers than the IRK

closed 2022-11-23 by stale[bot]

MehrCurry, 2023-11-14:
I have to come back to this issue. I am having the same problem. But i do not see any espresense/settings in the MQTT Explorer. I already delete the whole espresense topic tree but my enrolled device are still there. Did i missed something?