ESPresense/ESPresense #682: Add Apple IDs references to docs

Rahulsharma0810, 2022-09-17:
Its not a bug, I find it very difficult to get the real names of apple devices, from many devices in my Home. However adding it to docs may help some people to identify their devices. 

apple:1005:9-8 - iPhone (not sure which model)
apple:1005:9-12 - apple-tv
apple:1005:9-24  - apple:watch4-2 (Apple Watch Series 4 44mm case (GPS))
apple:1005:9-26 - apple:watch6-1 (Apple Watch Series 6 40mm case (GPS))
apple:1006:10-7  - apple:iphone14-5 (iPhone 13)
apple:1006:10-12 - apple:ipad11-1 (iPad mini 5th Gen (WiFi))
apple:1007:11-7 - apple:iphone14-5 (iPhone 13)

AudioAccesory5,1 : HomePod Mini (notsure about ID)

AlfredJKwack, 2022-09-17:
Those ID's are not stable across people and locations. They're 'best guesses' specific to you and your situation. There are enhancements on the way to make this better or at least easier.

However in the meantime you should be able to get this done relably as follows:

For your iPhones/iPad's: follow the steps in the docs for  [iOS Manual Setup](https://espresense.com/beacons/apple#ios-manual-setup)
1.  Go in your browser to the ESPresense devices page: http://espresense-office/ui
2. Fill in any name for your device in the `name` field and click the `Enroll` button. Be mindful that the name you give should preferably not contain spaces (see #312). The name you give will be used later and can by anything you like. For example you could call it `ble:His-Iphone13mini`
3. On your iPhone to to Settings -> Bluetooth and you’ll see a new ESPresense device.
4. Pair the device and the key becomes visible in the MQTT topic espresense/settings

At this point your iOS device will show up in the fingerprint list ( /ui/#/fingerprints ) with the arbitrary name you gave it.
If you look in [MQTT explorer](https://mqtt-explorer.com/) you should also see a topic `espresense/settings` underneath which there will be one or more child nodes starting with `irk:` followed by a long alphanumeric string. Underneath one of those that you will see a config node with the `id` you provided earlier. Using the example above it would be something like `{"id":"ble:His-Iphone13mini"}`. That same Id is then the one you would use in your HA setup where the mqtt_room sensor is concerned (ie. device_id: `"ble:His-Iphone13mini"`)

For your Apple watch: follow the steps in the docs for [WatchOS](https://espresense.com/beacons/apple#watch-os). That's a fair bit more involved unfortunately. I do wish we were able to follow a similar approach to the above using the [bluetooth pairing capabilities](https://support.apple.com/en-us/HT204218) of the watch.

ndbroadbent, 2022-09-25:
@AlfredJKwack Is iit possible to do this for Macbook Pro and Macbook Air? I've determined that my Macbook Pro ID is `apple:1005:9-24`, and my wife's Macbook Air ID is `apple:1006:10-12`.(Quite a few `apple:findmy` devices as well.)
I think I can use these IDs for now, but would like to know if there's a better way (out of curiosity, and for the future.)

 I tried the enroll flow but the ESPresense BT device doesn't show up for my Mac so I couldn't pair with it. I also got all the BluetoothLE and BluetoothLESync IRKs I could find out of Keychain access, but it doesn't look like any of those are related to my Mac.

Does a Mac also use IRKs? Or is there just some command I can run in the terminal to make it easier to discover? Thanks!

AlfredJKwack, 2022-11-24:
Hi @ndbroadbent. I haven't tried adding laptops to ESPresense.

ndbroadbent, 2022-11-24:
@AlfredJKwack Thanks for your reply! I’m still interested in this, so please let me know if there’s any way I can help. I would be interested to know if Apple MacBooks use IRKs, or if there’s another way to identify individual laptops. Actually I can install and run software on the laptops, so there’s probably a beacon program that can send out a Bluetooth signal that ESPresense can detect.

closed 2023-03-09 by stale[bot]

raptile, 2024-08-18:
with iOS 17+ this enroll option does not work anymore.
Anybody who knows how to get the IRK now?