ESPresense/ESPresense #686: Wifi and HA config screens show different values for: absorbtion and max distance

AlfredJKwack, 2022-09-18:
**Describe the bug**

There's s a discrepancy between the MQTT Device configuration and the native Wifi Configuration screens for _absorbtion_ and  _max distanace_. You can set values in Home Assistant and they are not reflected in the native Wifi Config. 

**Config**

- Version: v3.1.0 

**Hardware**

- Firmware flavor: default
- Device (be specific): D1 mini esp32 CH9102X 

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.
[image]

Zipties, 2022-09-18:
For me, I have to hit enter after inputting them in HA or they don't update

AlfredJKwack, 2022-09-18:
Thanks for the tip @Zipties. Tried it but saw no change in behaviour. The HA logbook does update. I have entries like: "_{...] Absorprion changed to 4.4 triggered by service number.set_value [...]_". Refreshing the ESPresence wifi config page does not reflect those values. I can however confirm that changing the value sin the wifi config has an effect (as you would expect).

Zipties, 2022-09-19:
Did you try restarting the esp after changing the values?

AlfredJKwack, 2022-09-19:
You don't have to restart the ESP when you do it though the Wifi config screen do you? Just hit save and it takes the input. It would defeat the purpose if you had to reboot the device for it to take the input from MQTT.

AlfredJKwack, 2022-11-24:
Bump the stale bot.

DTTerastar, 2022-12-18:
The wifi settings are done 100% at startup.  So even tho the value takes effect immediately via mqtt, you won't see it in wifi settings until you restart.  It's just a limitation of wifi settings.

closed 2023-04-03 by stale[bot]