ESPresense/ESPresense #691: Locked iPhone stops sending information to MQTT server

ReNeGaDe124, 2022-09-24:
I have successfully enrolled my iPhone in ESPresense with irk. It appears with the Alias I have selected. When the phone is unlocked and used it is sending the information to the MQTT server, but when I lock it it immediately stops updating distance in the Fingerprints menu and will appear as not_home in Home Assistant.

Has someone else encountered this issue? How have you resolved it? Thank you!

nickelnine, 2022-09-25:
I've encountered exactly the same thing with my iPhone 12 Pro. Keeping the auto lock setting to Never solves the not_home problem in HA but obviously drains the battery a lot faster so not the best solution but works for now.

begunfx, 2022-09-29:
My son's Iphone 13 no longer reports anything but "away".  Not sure if this is a regression as this was working perfectly in a recent build.

mrrobotnic, 2022-10-13:
Same issue here where the iPhone 13 when it locks it appears to stop transmitting and the device appears as "Not Home" even when it is.

fredamn76, 2022-10-29:
I have the same issue with iPhone and IOS 16.0.3

sufyan-motala, 2022-11-22:
I also have the same issue on my iPhone 13 Pro with IOS 16.1

begunfx, 2022-12-22:
Can someone please take a look at this?  I still cannot see my son's IPhone with ESPresense.  Thanks!

empaa, 2022-12-23:
I’m also tearing my hair out having this exact problem with an iPhone XR. It’s working flawlessly with other Apple devices we have around the house; iPhone 14 Pro Max, iPhone 8 Pro and Apple Watch Series 3. But that damn iPhone XR just disappears as soon as the screen locks up. I’ve been digging around every setting related to Bluetooth, location, find my, hands off functionality etc on the devices and made sure all devices are setup exactly the same but still no luck.

nickelnine, 2023-03-23:
Keep alive

Simonh98, 2023-03-29:
> I’m also tearing my hair out having this exact problem with an iPhone XR. It’s working flawlessly with other Apple devices we have around the house; iPhone 14 Pro Max, iPhone 8 Pro and Apple Watch [...]

Noticed same behaviour on my iPhone XR. Any new findings on this?

lxgenie, 2023-03-30:
I'm also seeing this same behaviour on iPhone 13 running iOS 16.3.1

Seems to be very similar issue to #574 

Would be great to get a resolution to this.

jjbrunton, 2023-04-01:
iPhone 11 same issue. Phones are away while they are locked so kinda defeats the point.

DTTerastar, 2023-05-26:
I still don't know why some iPhones do this.  My 12 doesn't.  You should all catalog what might make your device unique and compare.

nic-nicol, 2023-07-06:
Same issue. phone needs to be 'awake'
iPhone SE
iOS 15.7.7 (highest available for this phone)
HA companion app latest
latest ESPresence firmware

DTTerastar, 2023-07-08:
Is it in low power mode?

erikkt, 2023-07-25:
I am so disappointed. I tested ESPresense with my iPhone 14 pro, and it worked beyond my expectations. I was going to replace monitor script for all my presence automations in HA. After testing I ordered and installed 10 ESP32 base stations, and installed them around the house. Only to find out my wifes iphone 12 has this problem. Now it is useless and I have to revert to monitor script.

erikkt, 2023-07-26:
I am happy again. My wifes iphone 12 now sends beacon all the time.  I believe @DTTerastar has a point. I am not really sure what fixed it, but what I did was to create "family sharing" in iCloud, include her and share my position with her, and vice versa. After that, everything was ok.

lxgenie, 2023-08-12:
It would appear it's defeintely iCloud related.

After reading comments above regarding iCloud not doing enough 'stuff' I went through my iCloud settings and I had photo sync turned off as I use an alternative photo syncing solution. 

I've now enabled photo sync in my iCloud and the issues I was having with not transmitting when screen is off have been solved!

ReNeGaDe124, 2023-08-23:
> I am happy again. My wifes iphone 12 now sends beacon all the time. I believe @DTTerastar has a point. I am not really sure what fixed it, but what I did was to create "family sharing" in iCloud, include her and share my position with her, and vice versa. After that, everything was ok.

I have just enabled Family Sharing and this seems to have solved the issue. For the past 3 weeks my phone is sending data to ESPresense even when locked. Thank you for this solution!

Simonh98, 2023-09-16:
> I am happy again. My wifes iphone 12 now sends beacon all the time. I believe @DTTerastar has a point. I am not really sure what fixed it, but what I did was to create "family sharing" in iCloud, include her and share my position with her, and vice versa. After that, everything was ok.

Unfortunately, I cannot confirm.. it doesn't work for me even with this setup. Photo sync is not an option for me. Does anyone know another solution?

ReNeGaDe124, 2023-10-17:
This issue has reappeared for me with no apparent reason. I have tried disabling and reenabling Family Sharing which has solved the problem initially but no succes.

Has this reappeared for anybody else? Have you found any other solution? Thank you!

ReNeGaDe124, 2023-10-25:
I seem to have found a workaround. I installed the room-assistant app (https://apps.apple.com/us/app/room-assistant/id1538642237) on my phone and let it run in the background.

So far my phone has been sending BLE signal to espresense all the time for the past 5 days, even without reopening the room-assistant app since the first time.

Can anyone who was having problems try this to see if it also fixes the problem for you?

DTTerastar, 2023-11-05:
My wife and kids don't run the app and they always broadcast so I'm not sure if that's the correct cause and effect.

Kummallinen, 2023-11-25:
I'm seeing a variant of issue with 2 iPhones but not with a 3rd.

As far I as I can tell sync settings and sharing setting are the same on all 3, then main difference I can see are the other devices on the account.

All 3 iPhones have paired Apple Watches and while the watches are being worn the phone keeps broadcasting when locked. Once the paired watch is removed for charging 2 of iPhones stop broadcasting, but the other doesn't. For the one that keeps working there is also an iPad on the same Apple account that is within Bluetooth range.

DTTerastar, 2023-11-30:
Yeah apple has to be broadcasting for a "reason".  My suggestion is always to keep turning on the magic features like watch unlock and shared clipboard until it works.  Nobody has quite figured out the rhyme or reason yet.

robertgetch, 2023-12-31:
I'm not an iOS user by one of my roommates is. Does anyone have a suggestion for what she could turn on to make her phone broadcast consistently. Currently it only broadcasts when the screen is on. She does not have a smartwatch (and will not get one as she seems to hate them) and so the phone doesn't have anything connected to bluetooth usually, sometimes her ear pod things and the screen is usually on.

Is there a software/app she can run that might get it to stay awake? I'm guessing this isn't super widespread because most people have iWatches (particularly if they're gadget driven people) which would generally force it to talk. The device isn't in low power mode and she doesn't pay for iCloud (uses google photos for backup). She does not have a Mac so I'm not sure Universal Clipboard works because it has nothing to hand off too.

Quite frustrating really as it works fine if the screen is on.

robertgetch, 2024-01-01:
Sadly, Room Assistant was not able to force this iPhone 14 on iOS 17 to broadcast either (it dropped out <12 hours later). I checked the settings and theoretically R.A. was allowed to run in the background and seems setup fine. Handoff/Universal Clipboard is on, Photo Backup is on, all the various iOS features listed above we've tried, except family sharing (for a basic reason that all her immediate family and trusted friends use Androids). 

**Interestingly** if she is using her AirPods and the screen is off it stops broadcasting as well, I noticed this today while she was listening to a podcast with the Airpods in her head and playing; once the phone screen was off the iPhone stopped broadcasting again which makes me question if its a continuity issue as the phone was in a constant state of contact with the headphones.

closed 2024-07-14 by stale[bot]

TB-SE, 2024-07-31:
It's unfortunately still an issue. My girlfriends iPhone 13 Pro has the same issue.

yuriisk, 2024-12-09:
Have this issue as well. Nothing mentioned helped me. But I found that if I add any other apple device to iCloud account, the iPhone starts to advertise when screen is locked.
Does it worth to mention in docs?

DTTerastar, 2024-12-10:
Yeah, I always thought that was the issue... You're the first to really confirm that is the issue.  If you don't have anything else to do Continuity with it doesn't bother.

Chrunchy, 2026-09-02:
I've been using espresense for about 2years, with multiple iPhones, and so forth no issues with locked screens.
However after updating to iOS27 on one of phone (iphone 15pro), it stopped broadcasting unless my watch is on (i very rarely use the watch).

Eager to see if this is a "bug" related to iOS27, or if espresense will continue to work when iOS27 goes official :\
Quite dependent on espresense today, as locks, alarms, events etc. are used, or atleast was for now.

robertgetch, 2026-09-02:
It continues to be an issue for me with any iOS device that doesn’t have a
reason to do continuity for me.

*Robert Getch *
Cell: (503) 729-8258
***@***.***

Chrunchy, 2026-09-03:
Well, i'm not entirely sure anymore.

My device stops being seen by espresense with the IRK when the screen is off, but theres a rotating MAC with the same signalstrength that is always seen regardless of screen on or off.
Not sure what to make of it.