ESPresense/ESPresense #692: Calibration help

rjbutler, 2022-09-24:
I have 6 espresense devices in a 3 story townhouse.  They are ustbasically stacked in rooms directly above one another with 3 on each end of the house.  I'd estimate they are 3 or so meters apart on each end (with a ceiling/floor in between.  The devices on on the same floor are about 4m apart and there are walls between them.

I have been trying to tweak RSSI@1M and absorption to get them to accurately report the correct room in HA but it frequently happens that the ESP32 in the same room reports a higher distance than the device a floor below and they get mapped to the wrong room.

I have the RSSI@1m set for my iPhone at 1M with no obstructions (-54).  I have been trying to adjust the absorption of the devices in the rooms that are above/below each other to report correctly but  even when I think I have it, the next day it ends up reporting incorrectly.  Sometimes my wife is in the room above me and mine reports in her room and hers reports in mine.   

Does anyone have any guidance here? 

They're all the same brand ESP32 D1 Mini devices.

jerodjchastain, 2022-09-29:
Have you found any help on this?

rjbutler, 2022-09-29:
> Have you found any help on this?

I was mostly able to work around it by tuning absorption and then tweaking my automation so occupancy has to change for more than a minute.  The devices briefly get assigned to the wrong room pretty regularly but not for long.

jerodjchastain, 2022-09-29:
Yeah some of mine have been doing the same thing

jerodjchastain, 2022-09-29:
Would I need to go up or down with the absorption if I don’t want the esps to see devices in other rooms?

ethanholz, 2022-10-15:
I have also had issues with calibration. I would like to propose developing more documentation on calibration. Maybe a compilation of what people do to tweak this system to their needs?

OptimusGREEN, 2022-11-11:
Yeah I've been having a headache with these kind of issues recently too. More tutorials/docs would be welcomed.

ozett, 2023-01-03:
maybe a fresh issue is needed for more details on calibration?
no progess on this topic here in 6 month..

i placed today a ibeacon in **1m distance** to an ESPRESENCE device 
and got a **2m distance** reported.

what are the steps to correct this? 
i see changes in the RSSI and RSSI@1m value.

[and even after reading the docs ](https://espresense.com/calibration)i dont know exactly what to do for correction.
do i have to make changes for that beacon, or in general?
[does the buildin values](https://github.com/ESPresense/ESPresense/blob/master/lib/BleFingerprint/rssi.h) have to be checked as well for my indiviual location?

|sample, animated|
|:---:|
[image]

DTTerastar, 2023-01-03:
The reason antennas are always put high up is they work much better up there.  I am trying to figure out how to correct for this in my espresense companion app.  I have a scenario per floor that gets calculated via just the nodes you want and comes up with a confidence value for the scenario and we use that to pick the most likely scenario.  But it is still a WIP.  I'm trying to make it easier to calibrate by having the gui show the distances graphically with a built in edit.  Device editing works node settings are next.  It runs as an add on in haos.

ozett, 2023-01-04:
i am willing to experiment a bit,
but as far as i understand your post, there is no need for elaborate calibration, 
because it can not be extremly more accurate because of the hardware conditions anyway.

i read about the idea of zoning.
after rough triangulation, rely on only one device in that zone for distance values (RSSI). 
maybe adittionally use AI or a kalman Filter to smooth out the distance for RSSI
https://www.beaconzone.co.uk/blog/ibeacon-microlocation-accuracy/

but i looks thats all over my head, so i keep on following your efforts until i see a way to make use of it.
I planned espresence to help me open the front door if an ibeacon is closer than 50cm. but it doesn't seem all that easy to implement...

as for now, congratulations for all that has been achieved

closed 2023-04-18 by stale[bot]