ESPresense/ESPresense #697: Possible to allow mqtt_room to report multiple nearby sensors?

gourdo1, 2022-09-29:
**Is your feature request related to a problem? Please describe.**
I'm trying to improve the reliability of Espresense in a large room where I only want a trigger if a beacon is in a small area of the room. While setting absorption and max distance works some of the time, I find that throughout the day, distance reports can be wildly varied and obstacles like tables and chairs get in the way of accurate reporting. If I could trilaterate or at least use a pair of sensor devices, I feel that I could improve overall accuracy, but it seems that mqtt_room currently filters distance to the nearest sensor, even when the raw MQTT messages appear to be tracking reports from multiple sensors.

**Describe the solution you'd like**
Allow mqtt_room to report distances to multiple sensors found to be in range of a given beacon.

AlfredJKwack, 2022-10-02:
Hi @gourdo1 that's beyond the span of control of this project I believe. The `mqtt_room` feature of HomeAssistant controls what gets exposed. Looking through [the code for that](https://github.com/home-assistant/core/blob/dev/homeassistant/components/mqtt_room/sensor.py#L123-L134) it seems like it uses a combination of last seen and distance to decide which beacon gets reported where.

That being said there is an interesting project https://github.com/ESPresense/ad-espresense-ips which may suit your needs. Have you checked it out?

gourdo1, 2022-10-03:
Thanks! That makes sense. I didn't realize espresense wasn't responsible for mqtt_room integration. Turns out NodeRed has a node that reads raw MQTT data so I was able to get that working in the interim. I will check out the IPS project as that may be an even better solution.

closed 2022-10-03 by gourdo1