ESPresense/ESPresense #699: Apple Watch manually added IRK not showing at all

JeroenTuinstra, 2022-10-04:
Managed to retrieve the IRK code of my apple Watch - including decoding and reversing.
Added the 32 character long code to "Known BLE identity resolving keys" (without `irk:` in front of it).
Added the following code to HA: 

```
  - platform: mqtt_room
    device_id: "irk:###############################"
    name: 'ESP Apple Watch Jeroen'
    state_topic: 'espresense/devices/irk:###############################'
    timeout: 10
    away_timeout: 120
```
In MQTT explorer the IRK is not showing up anywhere. And off course in HA the Apple Watch is `not_home`. 
I have successfully added my iphone which shows up in MQTT Explorer as: 
```
espresence/settings/irk:################################/ config = {"id": "Iphone Jeroen"}
```
And my HA sensor works fine:
```
  - platform: mqtt_room
    device_id: "Iphone Jeroen"
    name: 'ESP Iphone Jeroen'
    state_topic: 'espresense/devices/Iphone Jeroen'
    timeout: 10
    away_timeout: 120
```
So why is the IRK that is mannually added under known BLE entities not showing up? 

**Config**

- Version: 3.1.0
- Active scan enabled: not possible to be set anymore
- Include filter: none
- Exclude filter: none
- Query filter: none

**Hardware**

- Firmware flavor: ESPresence Verwsion: v.3.1.0
- Device (be specific): ESP32-WROOM-32 (Wavehare e-Paper ESP32 Driver board)

JeroenTuinstra, 2022-10-04:
Turns out I used the wrong Base65 to HEX decoder. 
If you use this one: [https://base64.guru/converter/decode/hex](https://base64.guru/converter/decode/hex) and reverse the result, you get a different code than when you use the decoder from here: [https://espresense.com/beacons/apple](https://espresense.com/beacons/apple)

Can someone explain that to me? So I used the first link to decode base64 to hex with the following: 

```
9R8nI+Oy1dXtIiZE5FVJkQ==
```
The result was: 
```
f51f2723e3b2d5d5ed222644e4554991
```
Revers that manually: 
```
1994554e446222de5d5d2b3e3272f15f
```
Do the same thing in the second url and the result is: 
```
914955e4442622edd5d5b2e323271ff5
```
So what is going on here?

1liminal1, 2022-10-13:
So I'm jumping on this, have the exact same issue. Followed the guide, iPhone works fine, but the apple watch IRK does not. 

The only think I did differently was use the decoder in the guide. Did I miss something, it seems to say it, its a one stop shop

"Paste the contents in the form below and click ‘Decode’ to convert this into an IRK. Under the hood, this is extracting Base64 data and converting it to a hex string and then reversing the order of the bytes."

https://espresense.com/beacons/apple

Or did I miss something? 

Thanks all :)

JeroenTuinstra, 2022-10-13:
@1liminal1 So you don't get any result from the webpage? The easiest is to just paste in the code. Just check very carefully which code you take. It worked for me.

1liminal1, 2022-10-13:
Yup I did and pasted the code, but it wont pick up the watch :( 

Followed the guide to the letter. The MAC matched what was in key chain, did the convert on the website. Added the IRK to the web UI (just the string, no IRK prefix) then copied my phone senor (that works) changed some of the values of course lol and ... I've clearly done something wrong HAHA

Ill keep looking. I have the ultra on the latest version, I wonder if any changed.