ESPresense/ESPresense #700: Mini D1 ESP32 - extremely difficult to flash firmware

JeroenTuinstra, 2022-10-04:
**Describe the bug**

Using the firmware installation via the web - it is extremely difficult to flash the Mini D1 ESP32 board. I bought 5 and so far managed to flash 4 but with great difficulty.
When plugged in, it recognizes the com port and connects to the board. Yet when finally installing it ends with the error message: 

```
Installation failed

Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.
```
This keeps going the whole time. When I go to [https://esphome.github.io/esp-web-tools/](https://esphome.github.io/esp-web-tools/) it flashes find when I try the live demo. It flashes esphome right on it.
When I then go back to espresense.com it fails with the same error. 

Is there anything I am doing wrong, how can easier flash Mini D1 ESP32? And have others got the same issue?

**Config**

- Version: 3.1.0
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor: Verbose v.3.1.0
- Device (be specific): AZ Delivery Mini D1 ESP32 (ESP32-WROOM-32)

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

Clubsport, 2022-10-06:
You have to connect GPIO 0 to GND to bring it into flashing mode.

runarbjorn, 2022-10-08:
Hi there, I had the same problem. I found some online installer that works with these boards, can't remember the name of it. It was some kind of a led controller thing. I was also able to use esp flashing software to do it as well as the web flashing utility for bluetooth proxy. Sometimes I am able to use the esp presense installation tool but not always.

I had tried to put GPIO 0 to GND without success.

I also have a question for op about those devices that you were able to flash. Are they stable? I am getting problems with stability on my devices. They seem to lose connection to the wifi from time to time.

JeroenTuinstra, 2022-10-09:
@runarbjorn They seem to be stable at the moment. Off course their antenna's are not the biggest so a good wifi signal in the home is necessary. 

I managed to get them all 5 flashed by just trying and reconnecting and resetting them multiple times and suddenly they would initialize and upload the firmware. Strangely it works right away with the example of ESP webtools, where it installs a version of ESPhome. No initiation problems whatsoever. Also programming by Arduino IDE works fine. Just this webtool from espresense.com doesn't work - or with great difficulty.

Connecting GPIO 0 and GND didn't do the trick with me either. D1 documentation says that they go into flash mode automatically at startup. I guess they remain in flash mode for a while and quite that mode. Perhaps the speed of the website offering the web-tool has an influence on whether this time window of flash mode is reached. This could explain why it would work on 1 website and not the other.

armdan, 2022-10-17:
Interesting,  I have same boards and I flashed them multiple times with no problem by going to espresense.com  then install.
Hold down reset button till it starts flashing then let it go. I use this boards ( NOT an affiliate link  !) https://www.amazon.com/gp/product/B08PNWB81Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

nuk3man, 2022-10-27:
I'm getting the exact same error with the same card. However, ESPhome works just fine, exactly like OP reported.

kyledam, 2022-12-07:
I got the same issue when flashing via WebUI, I managed to make it work with ESPHome-Flasher, you can try it with manual flashing the espresense firmware from https://github.com/ESPresense/ESPresense/releases
ESPHome-Flasher can download from https://github.com/esphome/esphome-flasher

Stubbs, 2023-10-02:
I', still having the same issue with D1 mini's. I've not had 1 of them install yet.

avellma, 2023-11-12:
I did try to put GPIO 0 to GND and the reset button in the following order.
Step1: Connect the D1 Mini ESP32 to the PC
Step2: Go to the https://web.esphome.io/ page to connect 
Step3: Click connect, then press reset, then put GPIO 0 to GND, then release, then release reset

That work for me. Thanks

tome88, 2024-06-01:
Hi, I'm having the same issue re pressing boot button via web interface and even when trying to flash via esphometools. Where is the GPIO 0? Ive unscrewed the back of the device and see it on the USBC diagram but cant find it on the board?

Fruchuxs, 2024-11-18:
For anyone who hits this issue: You need to wire GND and GPI0; the two that are opposite each other