ESPresense/ESPresense #702: Spelling mistake in the project name - Should be "ESPresence"

ndbroadbent, 2022-10-05:
Apologies if this has been brought up before, but I searched through the GitHub issues and didn't find anything for "typo" or "spelling".

It's been bugging me for a little while that there's a spelling mistake in the project name. As far as I'm aware, I don't think "presense" is a valid spelling of the word "presence".

Here's another way to look at it: You get tons of results if you search for "presence" in the Home Assistant Core codebase:

[image]

And zero results if you search for "presense":

[image]

`espresence.com` is unavailable, but I'd be happy to pay for `espresence.io` or `espresence.org`, or some other TLD.

Perhaps this could be done for a version 3 major release.

Please feel free to close this issue if it's too late to change the name. Honestly I just wanted to open this issue to scratch my own itch!

ndbroadbent, 2022-10-05:
I forked the project and quickly renamed everything: https://github.com/ndbroadbent/ESPresence

markmghali, 2022-10-12:
ESP Pre Sense

ESPresense

closed 2022-10-12 by ndbroadbent

ilium007, 2022-12-08:
Came here with the same query!! https://www.askdifference.com/presense-vs-presence/