ESPresense/ESPresense #709: Set Static IP Address for ESPresense devices

woody166, 2022-10-11:
Hi there,

is there a way to set a static IP address for the ESPresense devices

AlfredJKwack, 2022-10-20:
Would your router not allow you do to that if needed. 

Also, if your use case it more related to finding the UI when IP changes, could you try typing in the name you gave your device followed by `.local` or `.localdomain` in your browser.  Altenatively, try typing `arp -a` in your shell could give you the name. You should be able to get something like `http://espresense-living-room.localdomain/`

Do these work?

Trevsweb, 2022-10-22:
I set the IPs on my router. worked fine but some of them were being a bit persistent with holding their ip address. took a few weeks and power cycles for them to reconnect. would be nice to be able to set it on the device

woody166, 2022-10-22:
Hiya, yeah I ended up setting it up on my router, which I had done before but was having issues with my router and had to replace it a couple of times and the reason behind asking the question if the IPs could be set on the actual devices

closed 2023-02-03 by stale[bot]

FARKIr, 2024-11-07:
I would also greatly appreciate the option to set a static IP for ESPresense nodes. This setup would provide more stability and control over network connections, especially in situations where DHCP server access is limited or for certain applications requiring a reliable network address. I believe this feature would be beneficial for many users who utilize ESPresense for home automation.