ESPresense/ESPresense #720: Feature implementation: Xiaomi Mi Flora plant sensor

0Flush0, 2022-10-23:
### Is your feature request related to a problem? Please describe.
I wanted to use my Xiaomi Mi Flora plant sensor  with espresence, but the project doesn't support it at the moment.
The sensor can read the soil's data like this (This would be sent to the MQTT server): 
{"id":"MiFlora: ------","mac":"--------","rssi":-57,"battery":100,"temperature":23.79"moisture":58,"light":113,"conductivity":4364}

### Describe the solution you'd like
I have **already implemented** the code needed into my local firmware, but I would like to contribute it to this project.
What my code does is actively connecting via Bluetooth to all active miflora devices, which the user has specified via bluetooth adresses in the ui. Then it reports the buffer, containing the data, to the MQTT server with a subtopic like this roomstopic/miflora/bluetooth-address.

### Addition context

I would really like to contribute to this project, and it would be nice hearing from you :)

Links for reference:
https://github.com/vrachieru/xiaomi-flower-care-api#protocol
https://github.com/basnijholt/miflora 

[image]

tpftpf, 2023-01-05:
This would be a great addon if implemented!
Tom

0Flush0, 2023-01-05:
Allready working on a pull request :)

SR-G, 2023-06-04:
What is the status of this feature request ?
It would really be nice to have this integrated back in ESPresence.

0Flush0, 2023-06-05:
The pull request is not going well, but you can download a version from it. I know this is not ideal and its not really up to date. If you want this to work now, i could  merge the new commits with my current version and help you get this setup.

SR-G, 2023-06-05:
Yeah, in fact (and maybe i have been a bit too quick on that topic...) i've just received a recently ordered whole bunch of M5 devices, with the hope to install ESPresence + MI FLORA inside them (right now i have a single linux box, but it's not working at all, due to current issues in the bluetooth stack on recent builds (generating freezes) + issues regarding the bluetooth range at home (generating connection errors))

So you mean, the best for me would be to (knowing i'm new to ESPresence) : 
- take the current master code from that repo
- reinject the change from PR #779 (a few "new files", main .h/main .cpp and BleFingerprint ?

Also, i've seen this, how is it related ? "Miflora fingerprinting" #781

0Flush0, 2023-06-05:
I can do that on the weekend and will get back to you. Also the Commit #781 added support to id the miflora devices using the bluetooth uuid. So this commit didnt really add much functionality!

0Flush0, 2023-08-09:
I finally updated the repo, sorry for  this long pause and I hope it still helps you. If  my pull request would go threw , so I'm not needed anymore to maintain this version.

SR-G, 2023-08-22:
Nice, i'll test this asap.

user34756361233, 2024-06-05:
I am just getting on board with ESPresence and have a number of these sensors. How can I activate them? Where can I find the "query" option?

Kind Regards,

user34756361233, 2024-06-05:
I have added the MiFlora sensors that I have under query. I however do not see the measurements in MQTT. How can I troubleshoot?