ESPresense/ESPresense #726: Garmin Forerunner 255 Music disabled when connected to phone

jazbraz, 2022-10-31:
I dont know if it is a bug or not, but when my garmin forerunner 255 music is connected to my iphone it stop sending out signals or theliver data again.

**Config**
  - platform: mqtt_room
    name: Garmin
    device_id: "xxxxxxxxxxx"
    state_topic: "espresense/rooms"
    timeout: 5
    away_timeout: 120

- Version:
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**
ESP 32 DEVKIT V1.

- Firmware flavor:
ESPresense 3.1
- Device (be specific):
Garmin Forerunner 255 Music
**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

j3ffrw, 2022-11-14:
This is a known limitation of tracking Garmin watches and some devices. I have a Garmin watch myself and it exhibits the same behaviour. https://github.com/ESPresense/ESPresense/issues/209#issuecomment-1039461938

ziolelle, 2022-11-29:
I confirm the same behaviour with a Garmin Forerunner 935... 
Really a pity... a smart watch like these would be the optimal choice for presence detection.

bogie, 2023-01-16:
I am still trying to find ways to make it work. Here is the discussion on the Garmin development forums: https://forums.garmin.com/developer/connect-iq/f/discussion/223399/ble-discoverablity-after-paired/1554403#1554403

closed 2023-04-30 by stale[bot]