ESPresense/ESPresense #727: M5Stamp C3 Mate (ESP32C3) - Distance Scaling Issue

Jeepmb, 2022-11-01:
**Describe the bug**

Experiencing an odd behavior from a batch of ESP32C3's regarding calculated distance. The reported distance in HA is only a fraction of what it should be, but does still scale. By my observations it will report a 0.05 M change for every 0.5 M actual distance change.

Measurements (in Meters):

Actual : Reported
0.5 : 0.30
1.0 : 0.35
1.5 : 0.40
2.0 : 0.45
2.5 : 0.50
3.0 : 0.55

I have been successfully using two ESP32WROOM boards with ESPresence up to this point. Out of the box, their distance measurements in HA were nearly spot-on with my two iBeacons (Galaxy Watch 5 and Galaxy S20). To expand my network, I bought 4 of the M5Stamp C3 Mate boards, as they were listed on the ESPresence web site. All 4 exibit this behavior. Playing with absorbtion factors would slightly change numbers, but not resolve the issue. I tried to flash previous firmware to test versions, but I was only able to get v3.1.0 to successfully flash via the web interface. I do see the note on the web site that ESP32C3 is in beta, but I am not sure as to what extent, as I see multiple posts describing regular functionality.

You can see in the logs a SHA-256 checksum error, but I am not sure if it is related.

**Config**

- Version: v3.1.0
- Active scan enabled: ?
- Include filter: None
- Exclude filter: None
- Query filter: None

**Hardware**

- Firmware flavor: esp32c3
- Device (be specific): M5Stamp C3 Mate v1.1 (ESP32C3)

**Logs**

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382134
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000
Setup network
Connecting to WiFi SSID '***********'.............................................................................192.168.1.***
Firmware:     esp32c3
Version:      v3.1.0
WiFi BSSID:   94:A6:7E:00:AA:C1 (channel=9 rssi=-40)
IP address:   192.168.1.***
DNS address:  192.168.1.***
Hostname:     espresense-office2
Room:         Office2
Mqtt server:  192.168.1.***:1883
Max Distance: 16.00
Init Free Mem:198364
PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=19 scl=18
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
```

febalci, 2022-12-06:
Worse in C3U mate; standing at 7 meters but reporting 0.67 meters. Like 1/10.

51av0sh, 2023-01-28:
Exact same issue. I replaced a dead D1 mini with a spare m5stamp i had laying aroundand I'm seeing the same issue reported here. I'm sitting 3m away and the distance reported is 0.7m. The D1 mini was usually around 2.9-3.1 at the same location

michaelblight, 2023-01-30:
It is very confusing - I have 5 m5stamp c3u's. Both my Apple Phone and Watch are sitting on my bed, and "br_2" running espresense is under the bed.

What I don't understand (for example) is for the watch the rssi is -51 for "br_1"and -44 for "br_2", but espresense thinks it is closer to the weaker signal. "br_1" is actually about 4 metres away on the floor above.

[image]

A bit later it settled on this below where it thinks the watch is in the kitchen rather than on the bed. The kitchen is the furthest room away in the house (>10m), and has the weakest signal (as expected), yet the distance is the second closest (why?), and the espresense picks it anyway. 

[image]

RolandWijnen, 2023-03-06:
I also have 5 m5stamp c3u's and 1 m5atom, with ESPresense installed and configured (leaving default settings as they are). I don't really get good results with detection. Similar to what @michaelblight described above. I've also installed ESPresense Companion and created a floorplan to better see what is going on. The locations of 3 iPhones and one Android phone reported there are also not correct and seem to differ from what the mqtt_room sensors are reporting. Have you figured out how to get the m5 base stations working? I'm using the latest beta release on these devices.

closed 2023-06-22 by stale[bot]