ESPresense/ESPresense #732: Configuring a radar sensor in v3.1.0 takes ESP32 offline

gith-account, 2022-11-27:
### Describe the bug

I use this ESP32 device:

https://www.amazon.de/diymore-Entwicklungsplatine-NodeMCU-Module-Bluetooth/dp/B09Z6RLS7R
https://raw.githubusercontent.com/AchimPieters/esp32-homekit-camera/master/Images/ESP32-38%20PIN-DEVBOARD.png

It works fine with v.3.1.0 unless an RCWL-0516 radar sensor is setup in the web UI.

https://www.electroschematics.com/wp-content/uploads/2017/08/1-RCWL-0516-Intro.png

The sensor is connected to 5V, GND and GPIO5 with the according settings:

PIR motion pin type: Pullup
Radar motion pin (-1 for disable): 5
Radar motion timeout (in seconds): 0.5 (default)

When this is setup in the web UI (same result with other GPIO pins) and the ESP32 device is restarted, the ESP32 device becomes inaccessible: the web UI is unavailable, the device does not respond to pings, the connection to the MQTT server fails:

Disconnected from MQTT; reason 0
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
(...)

The ESP32 device is accessible via USB and the console with huge delays, seems to connect to the WiFi network, sees BLE beacons, and reports motion events.

The same setup works in v3.0.0.

### Version

v3.1.0

### Include filter

3 BLE beacons

### Firmware flavor

Standard

### Device (be specific)

diymore ESP32 NodeMCU CP2102

gith-account, 2022-11-28:
Oh no, should the PIN type be Pulldown? I will try tomorrow.

peter-dolkens, 2023-03-03:
I had something like this, but not 100% sure what yet.

I've actually got my prototype working at the moment, and continue to monitor.

Will write back here if I manage to get a replication.

closed 2023-06-22 by stale[bot]