ESPresense/ESPresense #738: Name field not available to enroll iOS device

beyondwatts, 2022-12-01:
### Describe the bug

Following instructions @ https://espresense.com/beacons/apple

>     Go in your browser to the ESPresense devices page: http://espresense-office/ui
>     Fill in a name for your device in the name field and click the Enroll button.
>     On your iPhone to to Settings -> Bluetooth and you’ll see a new ESPresense device.
>     Pair the device and the key becomes visible in the MQTT topic espresense/settings

Using v3.1.1, there does not appear to be a field to enter a name value at http://[esp-ip]/ui only titles "Devices" and "Fingerprints". Clicking on "fingerprints" shows IDs for devices publishing to mqtt

Am I missing something?

### Version

 v3.1.1

### Device (be specific)

eps32-c3-DevKitC-02

### Screenshots

[image]

beyondwatts, 2023-01-10:
I managed to get this working with a different chip...

The failing model was: ESP32-C3-DevKitC-02 is an entry-level development board based on ESP32-C3-WROOM-02
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html

The working model is: DUBEUYEW ESP32 based on ESP-WROOM-32
https://www.amazon.co.uk/dp/B08CCYWZN3

What information would be useful to debug the failing chip?

michaelblight, 2023-01-29:
It took me way too long to realise I'm getting the same issue with an **m5stamp** (the C3U version). To be clear, going to the "my-ip/ui" page shows "Devices", but the detail page is completely empty. The Fingerprints page appears to work fine.

jadenv, 2023-02-15:
I'm using a M5Stamp C3 Mate and got stumped on this as well. Glad I stumbled across your suggestion @michaelblight, thank you!

shane64, 2023-03-28:
Same issue here with an ESP32-C3-12F-Kit.  Thanks.

closed 2023-07-11 by stale[bot]