ESPresense/ESPresense #744: BH1750 update frequency

simonyel, 2022-12-04:
I have BH1750 light sensors configured and working. However, the update frequency is over a minute. 

This throws off my light automations: light cuts off because no presence is detected, and the count is below 1. Less than a minute later, someone else walks in, and the lights fail to turn on because the light sensor is still showing plenty of light present.

Can the update frequency be set to something usable, similar to the DHT22, please?

simonyel, 2022-12-05:
> @simonyel<https://github.com/simonyel> I raised the same issue here: #667<https://github.com/ESPresense/ESPresense/pull/667>
>
> They "fixed" BH1750 when there were actually no issues... Still the same for a few months, the new update still has super slow update frequency. Messed up my automation... My ESP doesn't have USB, so I need to remove all my ESPresence and change back to ESPHome when I have the time to do it.

Wasn’t there an “update from URL” field? I’m thinking of using that to shift back to ESPhome as well.

MiningPickaxe, 2022-12-18:
This is really annoying for me as well... may delay some actions by 60 seconds.
You can always just clone the repo yourself and revert https://github.com/ESPresense/ESPresense/pull/667 but that is not really the elegant and future proof way to do it.
I agree with the PR Author about reducing "spam" via MQTT, but I think a configurable delay would have been the wiser choice here (and also keep the old code that prevented it from sending a mqtt message if the value did not change).

closed 2023-04-03 by stale[bot]

simonyel, 2023-04-03:
wontfix. Well, then, you can't use this solution for intelligent light control.