ESPresense/ESPresense #748: Out of Memory - Reboot

jmcclure0921, 2022-12-05:
### Describe the bug

I have some device nearby that rapidly rolls its MAC address.  I can see the available memory on all of my espresense devices slowly dwindle to nothing and then reboot.   It takes approximately 2 minutes to do so.

A way to ignore a device by ID, ignore all rolling MAC address devices, or to select all of the devices that are monitored could be a workaround to this issue, 

### Version

2.2.9/3.1.1

### Active scan enabled

No

### Include filter

No

### Exclude filter

No

### Query filter

No

### Firmware flavor

Standard and Verbose. 

### Device (be specific)

ESP32 Mini NodeMCU

### Logs

```text
1 New   | MAC: 205caab2a8a5, ID: sd:0xfef1                                                 -88dBm 
1 New   | MAC: 1daddf976198, ID: sd:0xfef1                                                 -77dBm 
1 New   | MAC: 26d714def74d, ID: sd:0xfef1                                                 -84dBm 
1 New   | MAC: 2c67aaad8b43, ID: sd:0xfef1                                                 -69dBm 
1 New   | MAC: 3c34b8a90f5b, ID: sd:0xfef1                                                 -67dBm 
1 New   | MAC: 1f7b61b1d426, ID: sd:0xfef1                                                 -70dBm 
1 New   | MAC: 1ada2ab09b64, ID: sd:0xfef1                                                 -64dBm 
1 New   | MAC: 8dafc1e4164c, ID: sd:0xfef1                                                 -69dBm 
1 New   | MAC: 2f9bccd94455, ID: sd:0xfef1                                                 -70dBm 
1 New   | MAC: 237a50814087, ID: sd:0xfef1                                                 -67dBm 
1 New   | MAC: 186d88b65c7b, ID: sd:0xfef1                                                 -67dBm 

abort() was called at PC 0x401a00a2 on core 1

Backtrace:0x400831dd:0x3ffdf2700x40095361:0x3ffdf290 0x4009a309:0x3ffdf2b0 0x401a00a2:0x3ffdf330 0x4019d413:0x3ffdf350 0x40185a36:0x3ffdf370 0x400d92a4:0x3ffdf390 0x400d7c9e:0x3ffdf3d0 0x400d5d60:0x3ffdf400 0x400d5e60:0x3ffdf480
```

RyanAHolland, 2023-02-22:
I am in the same place. None of my m5atoms will work reliably even with the query filtering turned on.

cdenfert, 2023-03-21:
I have the same problem. Some of my devices that are intercepting a lot of blue traffic from my neighbors are constantly rebooting. Even if I only filter my iBeacons with the filtering option.

DTTerastar, 2023-03-22:
Yeah, c++ is a tough language.  At the moment every mac is the same size object (and it's not super small).  We'd need to keep a separate list of filtered macs.

DTTerastar, 2023-03-22:
The other would be to try and use a wrover instead of a wroom since those have more memory.

jmcclure0921, 2023-03-22:
@DTTerastar Won't work for me given how frequently it reboots.  Do you know the general area where the I assume malloc(), happens? I am pretty experienced with C++, just don't have time to dig through it to figure out how it all works. I can see if  I can come up with something that would work.

DTTerastar, 2023-03-22:
https://github.com/ESPresense/ESPresense/blob/858200e74570737131971936afa40bf1ad3b8ad8/lib/BleFingerprint/BleFingerprintCollection.cpp#L193

jmcclure0921, 2023-03-22:
I see a few reasonable options:

1. After the BleFingerprint is created, check if the "include" is not blank and includes its identifier, or if it is included on the exclude list.  If it is just delete it right then
2. You already do a cleanup for old items when copy is called, do the same cleanup for stuff excluded/not included.
3. You check for low memory already, if memory is low just stop adding new ones until some expire. I will note that my particular device never produced the low memory error, so it is possible that by copying the entire fingerprint vector, and not handling exceptions you are actually blowing the heap here. Should probably handle the exceptions, or check the available size of the heap against the size of the vector, or use a mutex to block the other thread so that the BleFingerprint vector can be used directly. 

A combination of 1 and 3 is probably optimal. I'll try and work out a PR when I have some free time.

DTTerastar, 2023-03-22:
That mac will just come back in 1000-3000ms... You can't just delete it because you'll have to redo all the logic to figure out what it is.  Exceptions are turned off, so it can't handle them... https://h2zero.github.io/NimBLE-Arduino there might be a way of filtering them out at this level and then they'll get expired.

closed 2023-07-09 by stale[bot]