ESPresense/ESPresense #749: Refuse to connect to network when not connected to laptop

NeoFax75, 2022-12-11:
### Describe the bug

When I originally flashed 5 ESP32-S3s with ESPresense, they connected fine and showed up in HA.  However, when I connect them to 5v connections in the room I assigned them, they refuse to connect to the network.  I put them back on the laptop and they connect to the network fine.  I don't know how to troubleshoot this phenomena.

### Version

3.3.1

### Device (be specific)

ESP-WROOM-32 https://www.amazon.com/dp/B0B19DXFHX?psc=1&ref=ppx_yo2ov_dt_b_product_details

ShonP40, 2022-12-23:
Are they in a reasonable distance from your router?
And can you check if your power supply is actually outputting 5V?

NeoFax75, 2022-12-23:
The closest to the router is appx. 12" from the router.  The furthest is
appx. 60'.  I checked the power and they are all 5v.  I purchased a
Huzzah32 and it connects way better, so I purchased 2 more and will
complete the remainder after the new year.  I think maybe the biggest
problem is there is like 30+ 2.4GHz wifi routers in my neighborhood that
overlap.  I have set my channels to deconflict and that has helped with
some of the non-Huzzah32 SBM.

ShonP40, 2022-12-23:
I’m personally using ESP32 WeMos D1 Minis & don’t have any disconnection issues with them

NeoFax75, 2022-12-23:
Here are the ones that are having problems connecting;
https://www.amazon.com/dp/B0B19DXFHX?psc=1&ref=ppx_yo2ov_dt_b_product_details

ShonP40, 2022-12-23:
Can you install ESPHome on them with the logger set to verbose and check if they still have WiFi issues?

NeoFax75, 2022-12-24:
Yes, they do.  I tried ESPHome as well and tried to get BLE tagging setup
that way.  They have the same problem.  I also bought a new TP-LINK AX55
router which helps as now two-three connect for appx. 3-4 hrs and the one
in the Master Bedroom stays connected for 24 hrs.

ShonP40, 2022-12-24:
Well so it just seems like a WiFi issue with the other ESPs you have
They probably have weaker antennas that are more susceptible to interference

Can't really help you with it that much

ShonP40, 2022-12-24:
You should also check each ESP's WiFi RSSI

NeoFax75, 2022-12-24:
Ok, I'll check it out.

closed 2023-04-07 by stale[bot]