ESPresense/ESPresense #750: [Q] Was the "active scan" switch removed?

Nihvel, 2022-12-13:
### Is your feature request related to a problem? Please describe.

I haven't used this project in a long time. Personal reasons. I remember there used to be a switch for toggling the scanning, and the proof is at this link https://espresense.com/home_assistant where the screenshot shows Active Scan.

I recently re-flashed one of the ESP32 i used to use time ago for this same ESPresense project. This time it shows these entities in my screenshot and no switch.

[image]

### Describe the solution you'd like

Highly appreciated to have the switch back. For as good as the project is, reporting is only 99% accurate, and it honestly is a waste of cpu to continuously sniff for beacons.

In Home assistant, automations can chose when to enable the scan. 

```
automation:
- id: scan when sensor detects movement
  trigger:
  - platform: state
    entity_id: binary_sensor.movement
    to: 'on'
  condition: []
  action:
  - service: switch.turn_on
    entity_id: switch.espresense_active_scan
  - delay: "00:00:30"
  - service: switch.turn_off
    entity_id: switch.espresense_active_scan
```

This will allow the ESP32 to "take a break" from scanning, and will only send the message once.
Accuracy could potentially not be 100%. This depends by the position of the ESP32 and vicinity with others. But it is higher than 99% because it does not continuously listen for beacons, causing a **False Negative** to be recorded.

The grey lines are the False Negatives.

[image]

DTTerastar, 2023-01-01:
Active scan doesn't add any real value to ESPresense and it adds lots of btle noise that detracts from the core location finding, so yeah, it's not recommended anymore.

Nihvel, 2023-01-02:
Well, if that's how it is now, fine! But how would it work in case I want to pause the scanning? Thanks!

closed 2023-04-17 by stale[bot]