ESPresense/ESPresense #751: Use dedicated integration in Home Assistant for convenience.

formatBCE, 2022-12-18:
### Is your feature request related to a problem? Please describe.

Hello!
First of all, thanks for your work - it really helps to ton of people!
However, to me, entry level for such project is way high.
One would need to have non-entry level of knowledge in MQTT, HA YAML configuration, and a lot of work to make everything useable. Adding new beacon requires manual changes in configuration, as well as scrolling through serial log of BLE node.

### Describe the solution you'd like

I tried your project about a year ago, and wasn't satisfied with several things in ESPresence, like:
- complexity of configuration and maintenance;
- recognition speed;
- lack of device_tracker object, that I could bind to person in HA.
So I created my own pet-project, which works great for myself. It doesn't have most of ESPresence features, for sure. 
However, here's main points that are useful for me:
- fast off-to-on time (literally, I'm coming out the elevator, stepping to my apartment door, and my lock is unlocking for me);
- ability to configure node through ESPHome (thus, ability to use this ESP device for other things in parallel, and easy management);
- dedicated integration in HA, that makes beacon management really easy, and allows some additional functionality (e.g. I use Kalman filtering for signal, and create proper device_tracker entities for each beacon).

Probably, you could consider something similar for your project in the future? I'd rather use someone's great solution instead of my own, as I'm not really good C++ developer...
You may check my repo https://github.com/formatBCE/Format-BLE-Tracker, and I'm ready to help you with any questions.

Thank you!

formatBCE, 2023-03-19:
Not stale.
Does somebody look into issues at all?

scaytrase, 2023-03-27:
I've tried it today, no clue how it's connects to ESPresense. I think it would be great if it gives us a list of ESPresence detected devices and create them by id (as ESPresense manual suggest).
I'm not sure adding my MAC is valid option since some devices may randomize their MACs, so beacon id (which is not always UUID in ESPresence) would be better

formatBCE, 2023-03-28:
> I've tried it today, no clue how it's connects to ESPresense. I think it would be great if it gives us a list of ESPresence detected devices and create them by id (as ESPresense manual suggest).
> I'm not sure adding my MAC is valid option since some devices may randomize their MACs, so beacon id (which is not always UUID in ESPresence) would be better
> 
> Also tried add my mi band by mac - not being tracked using this integration tracker. I hope it because it lack `mifit:` prefix to identify topic

My repo is nothing to do with ESPresence itself - it's my own take on BLE presence, that does work better with HA via dedicated integration, and uses ESPHome for scanning nodes firmware.

I added proximity UUID and MAC beacons there - it's only things I use at home.

formatBCE, 2023-06-26:
Not stale.

formatBCE, 2023-09-24:
Not stale