ESPresense/ESPresense #753: Issue with discovering multiple iphones

konsulten, 2022-12-18:
### Describe the bug

While one of the iPhones in the house consistently is discovered and tracked(and never lost even during night time when screen has been locked for a long time). However the other of the same exact type and firmware does not when screen is locked. Is there some limitations in discovery of multiple iPhones in same area?
I read something about this: 
https://uynguyen.github.io/2020/08/23/Best-practice-Advanced-BLE-scanning-process-on-iOS/ and this:
https://developer.apple.com/forums/thread/652592

Could this be related? That iPhone BLE works differently depending on some state? And might prohibit it from being discovered when in locked screen?

Also but a bit unrelated, I have not been able to find my apple watch using IRK from cloud ring( have published the irk config to mqtt ). The entry in the ring is named: BluetoothLESync. If that is of some help.

### Device (be specific)

2 x iPhone SE (second gen) and Apple Watch 4

konsulten, 2023-02-10:
One thought I got was that one phone was connected to an apple watch. And this phone is the one has reliant bluetooth tracking. So it might be it wont disable tracking when screen is disabled if it has a connected device.

b1ggi, 2023-02-20:
> One thought I got was that one phone was connected to an apple watch. And this phone is the one has reliant bluetooth tracking. So it might be it wont disable tracking when screen is disabled if it has a connected device.

can confirm, exact same behavior

closed 2023-06-08 by stale[bot]