ESPresense/ESPresense #754: MQTT not being sent for a couple of devices.

jonwilliams84, 2022-12-21:
### Describe the bug

Got 8 base stations set up around the house and for the most part all working really well. Noticed my wife's phone (Huawei P20 Pro) and my boys tablet (Nexus 7 2013), that were originally working fine started showing as 'not_home'.  

Upon investigation it seems as though no MQTT messages are being sent for these devices.  I have tried removing the exclude filter, but this hasn't made a difference. 

The devices are being picked up by 'Locate' Android app, and if I hop into the logs of the ESP32 I can see them.

Any ideas? I am at a loss as to why these devices have both suddenly stopped being sent via MQTT.

### Version

3.1.0

### Active scan enabled

no

### Exclude filter

msft:cdp:0922 msft:cdp:0902 apple:1005:9-5 md:038a:8 245a4c78d8d9 b4e3f942df96 name:jabra-evolve2-65 ad:267f0001-eb15-43f5-94c3-67d2221188f7 sd:0xfe9f apple:

### Device (be specific)

Huawei P20 Pro, Nexus 7

### Screenshots

[image]

[image]

totalitarian, 2022-12-29:
Same issue here. I can see the device in /ui/#/fingerprints but not in MQTT explorer

DoraDeu, 2022-12-30:
Same for me with a new iBeacon. The fingerprints are updating with the iBeacon ID but the device is not created in the object tree in ioBroker.

totalitarian, 2022-12-30:
> Same for me with a new iBeacon. The fingerprints are updating with the iBeacon ID but the device is not created in the object tree in ioBroker.

Check your max reporting distance, mine was set too low...

jonwilliams84, 2022-12-30:
> > Same for me with a new iBeacon. The fingerprints are updating with the iBeacon ID but the device is not created in the object tree in ioBroker.
> 
> Check your max reporting distance, mine was set too low...

I tried this, raised to 16m, even <1m away the devices still don't show in MQTT explorer. Its annoying because 4 other devices are working flawlessly. Just appears to be my son's Samsung tablet and wife's P20 Pro. No matter what I do, they just never appear in MQTT.

DoraDeu, 2022-12-31:
Valid point but this was not my problem. The distances have been under 16m.
I had to reconfigure my MQTT broker (uncheck "Publish only on change"). Only after this change the objects have been (re-)created. I am not sure if this is a bug on ESPresence side or "by design".

DTTerastar, 2022-12-31:
RSSI of -90 is super weak.  You gotta get the RSSI higher than that.  If using ha companion increase the signal strength.

rouloy, 2023-02-09:
Got a similar issue, but it seems to be a problem with the base station. One is sending out MQTT messages just fine even if the RSSI is quite low. Others (same brand) don't send out any MQTT message or only once after I plugged them in (reboot doesn't have this effect) but after couple of minutes they stop publishing even though the device sits right next to them

closed 2023-05-24 by stale[bot]