ESPresense/ESPresense #757: RCWL-0516 Doppler Microwave sensor False positives

Rahulsharma0810, 2022-12-21:
### Describe the bug

I am trying to make the Microwave doppler sensor work with Espresense, But it gives many false positives every few seconds.

Although I tried it in another esp32 board with simple code to test it, it worked fine.  From some online forums, I tried Ferrite with GND and 3v3 it didn't help.

Has someone tried to test the Radar Sensor with the ESPresense? I hope using it with ambient and Lux sensors don't interfere with its working. 

### Version

3.1.0

### Active scan enabled

yes

### Include filter

N/A

### Exclude filter

N/A

### Query filter

N/A

### Device (be specific)

ESP32

### Screenshots

[image]

[image]

DTTerastar, 2022-12-31:
Do you have a ceiling fan?

closed 2023-04-14 by stale[bot]

sebastienvg, 2023-09-01:
> ### Describe the bug
>
> I am trying to make the Microwave doppler sensor work with Espresense, But it gives many false positives every few seconds.
>
> Although I tried it in another esp32 board with [...]

I am having the same issue, did you ever solve it?