ESPresense/ESPresense #758: Is it possible to combine Espresense with WLED on the same board?

sirs2k, 2022-12-22:
### Is your feature request related to a problem? Please describe.

Hi all.

The title explains it all, would like to combine WLED with espresense on the same board instead of having multiple ESPs laying around everywhere 

Thank you

ShonP40, 2022-12-23:
Don’t really think that you can do that

There’s not enough RAM & flash storage on an ESP to have both running at the same time

Not to mention that some WLED effects require a lot of processing power, so sharing the CPU with another program will most likely make it crash

sirs2k, 2022-12-24:
Yeah makes total sense.
Through it would be interesting to test if it's at all possible. Even if it doesn't handle running both, then it would be up to the user to separate the 2 across separate devices or not

For someone like me, I'm only running very basic "ambient" animations through wled so I wouldn't think the esp can't handle both.

Having the ability to run both (or espresense as well as esphome) on the same MCU would be a game changer.

ShonP40, 2022-12-24:
There have been talks about ESPHome integration [here](https://github.com/ESPresense/ESPresense/issues/73) but it’s not going to happen due to ESP32 hardware limitations

Same goes for WLED as well probably

Your best bet would be to implement the features you need from WLED into a custom ESPresense build

closed 2023-04-08 by stale[bot]