ESPresense/ESPresense #760: how to retrieve the IRK ?

ozett, 2022-12-24:
### Is your feature request related to a problem? Please describe.

i used the new "enroll" feature, ([v3.1.3 enroll](https://espresense.com/beacons/apple#ios-manual-setup))
now the devices shows up with  its enrolled name.
but how to i retrieve the IRK ? 

the mqtt-topic /settings/ does not annouce the IRK,
also the /device/ topic does not show the IRK, only the name

### Describe the solution you'd like

IRK in the gui? 
on http://espresence-IP/ui ?

### Describe alternatives you've considered

do the enroll again and examine the /settings/ topic for catching the IRK,

jlambert121, 2022-12-24:
After enrolling my phone I have a `settings/<irk>/config` key with the `irk` being part of that path.  I haven't found I needed the `irk` though after it was registered either.

ozett, 2022-12-25:
to keep track of my data/devices i thought it would be helpful to reveal that data sometime from the espresence board.
the new gui would be a fantastic place to show such data, i guess  💭 
or maybe some mqtt-command to re-send the IRK from the storage on the ESP

closed 2023-04-08 by stale[bot]