ESPresense/ESPresense #771: The possibility to calibrate the rssi@1m for each beacon individually

voicubabiciu, 2022-12-28:
### Is your feature request related to a problem? Please describe.

Right now I have successfully configured two esp32-DevKitC modules with ESPresense v3.1.1 and I configured two beacons one is an iPhone 13 pro-Max and another one is an Apple watch series 7. The problem I am facing is when both devices are at 1 meter distance from the ESP module they show different distances. For example:
[image]

DTTerastar, 2022-12-28:
You can do it today.  It's a bit more manual than I'd like but you send a retained message to mqtt:

`espresense/settings/{id}`

{"name":"Darrell's Watch","rssi@1m":"-70"}

voicubabiciu, 2022-12-28:
Thanks, I'll try this right now and I will come back with an update

voicubabiciu, 2022-12-28:
Probably I didn't understand well so correct me if I am wrong. The id that should be sent in the mqtt topic `espresense/settings/{id}` is the id of the beacon (the IRK key) and the name from the payload can be anything?

voicubabiciu, 2022-12-28:
Ok, so I tried setting the `rssi@1m` on the following topics and none of them worked `espresense/settings/{id}` and `espresense/devices/{id}/office` where id is the one displayed in the fingerprints table of the ESP node. Also, I didn't find any documentation on this subject anywhere, and it looks like the doc is a little bit outdated.

voicubabiciu, 2022-12-29:
After a good sleep and after looking a little bit in the code on master it looks like the correct topic is `espresense/settings/{id}/config` and the payload should look something like this
`{"name": "Beacon mane", "rssi@1m": "-70"}`.

Also, it would be very nice to have the possibility to set these settings from the UI and also updated documentation for MQTT. Thanks a lot for your help @DTTerastar, your answer guided me to the correct solution!

closed 2022-12-29 by voicubabiciu

skylarv, 2023-03-26:
@DTTerastar I was able to get custom rssi@1m per beacon via publishing the settings using MQTT above and it's working. However, I noticed hitting the esppresense website URL, the device's settings is then reset (removing the custom rssi@1m set for that device so it goes back to default).

Is there a way to prevent it from resetting when navigating to the web ui? Thanks!

arcnmx, 2023-05-16:
> I was able to get custom rssi@1m per beacon via publishing the settings using MQTT above and it's working. However, I noticed hitting the espresense website URL, the device's settings is then reset (it removes the custom rssi@1m already set for that device so it goes back to default).

can confirm, it works on publishing the settings, but then does not load them when espresense resets - which it does very frequently if you try to access it over http (no idea if that's a crash or OOM or what, but that's a separate issue).

however, if I publish the new settings and then hit the Reset button in home-assistant, it seems to sometimes save it, if I'm lucky... it takes a few restarts of all espresense nodes to get them all to agree though :disappointed:

DTTerastar, 2023-05-25:
The settings should reload at startup, that's the whole point.  There was a bug in the older versions if packets got split up but AFAIK that doesn't happen anymore.  Make sure you allow retained topics on your mqtt server.  If that is disabled then it will won't retain the settings...

arcnmx, 2023-05-26:
> The settings should reload at startup, that's the whole point. Make sure you allow retained topics on your mqtt server. If that is disabled then it will won't retain the settings...

I can verify that the `settings/XXX/config` topic is being retained correctly - they're definitely getting lost on some resets despite that. IIRC the `rssi@1m` has trouble applying, but other config settings like `id`/name/etc are less of an issue. It's like a roulette, 75% of the time it'll come online with the wrong value, but sometimes it will work for a few minutes.

but... the bigger problem is that even without resetting the node, it will just forget the setting within a few minutes. I've had to set up a watchdog to monitor the `devices/XXX/room` topic and re-broadcast the settings whenever the `rssi@1m` reverts to the wrong value.
Here's it happening in realtime - notice how it just randomly starts to "forget" the rssi@1m, even bouncing back and forth until it settles back on the original/wrong value:
[image]

The node's uptime is 241000s, so it's been operational the whole time and no new `settings/XXX/config` payloads have been pushed. The moment I re-publish the settings (without changing the payload from what it last retained), it will work properly again for another 10 minutes or so.

DTTerastar, 2023-05-26:
Is it an iBeacon?  Those have an embedded rssi@1m in the broadcast and that code might conflict.

arcnmx, 2023-05-29:
> Is it an iBeacon? Those have an embedded rssi@1m in the broadcast and that code might conflict.

I'm pretty sure it is, yeah.

codemunkie15, 2023-09-05:
Is this still an issue? I am interested in setting this up.

DTTerastar, 2023-09-05:
I've checked the code I don't think this is still an issue