ESPresense/ESPresense #773: findmy-messages for fingerprinting?

ozett, 2022-12-29:
### Is your feature request related to a problem? Please describe.

i have some apple-devices sending lots of findmy-messages.
that messages should inlcude the unique-pub-key of the the device/owner.
starting with the 4th byte -> https://github.com/furiousMAC/continuity/blob/master/messages/findmy.md
but i assume the key is unique. (for each device?)

so anybody considered that information to use also in fingerprinting?

### Describe the solution you'd like

see another step forward in indentifying apple-devices with BT adverisments

### Describe alternatives you've considered

investigate this long story myself, 
but as normal i only want to provide a single piece of guessing..

DTTerastar, 2022-12-29:
That key gets regenerated every 24 hours, so it's usefulness is limited.

ozett, 2022-12-30:
what about using some fields from nearby-info and nearby-action ?
https://github.com/furiousMAC/continuity/blob/master/messages/nearby_action.md

the linked paper are very interesting, but maybe there are newer information and some more fields to use than for IOS12 ?

DTTerastar, 2022-12-31:
Nearby Info is exactly how we track iPhones and Apple Watches.

ozett, 2022-12-31:
i tried to make use of [the data-fields for the nearby-info](https://github.com/furiousMAC/continuity/blob/master/messages/findmy.md) with node-red,
and found in the end that substring comparison was sufficient, 
but with no great success i.e. in identifying phones in active calling-state.

https://github.com/1technophile/OpenMQTTGateway/issues/1372

i would love to see esppresence be configurable for generally sending out the manufacterer-data-string,
in somehow the [same way as  openmqttgateway.com is doing](https://docs.openmqttgateway.com/use/ble.html) it. but i dont see such option.

maybe now i look into your code here in the github to find some new ideas

closed 2023-04-14 by stale[bot]