ESPresense/ESPresense #777: Android is extremely slow

LuxStatera, 2022-12-30:
### Describe the bug

I am using the Home Assistant companion app to handle the ibeacon for Android. I can currently see on MQTT Explorer that it has sent 259 messages and this will increase by 1 every 5 minutes or so. My iPhone was setup with its irk and currently has 6575 messages and rapidly increasing every second. iPhone tracking has been perfect but Android takes way too long to update. I could enter every room in the house then go back to my room and it will still say that I am in my room that whole time because it did not update fast enough. There are several videos that show different ways to set up Android phones but only 1 method worked and documentation is sparse so I'm not sure what to. This is my first time messing around with MQTT. How do I fix this?

### Version

v3.1.1

DTTerastar, 2022-12-30:
The UI for the Android iBeacon stuff is very buggy.  If you click around enough you can set the frequency of sending faster.  But for me it took at least 2-3 times in the UI for it to actually do it.

LuxStatera, 2022-12-30:
Which settings did you change?

DTTerastar, 2022-12-30:
https://companion.home-assistant.io/docs/core/sensors#bluetooth-sensors

LuxStatera, 2022-12-30:
I had the transmitter power set to high and advertise mode set to low latency. Should I adjust those or change something else?

DTTerastar, 2022-12-31:
Low Latency is the correct option, but if HA Companion isn't doing that, set it to another option, restart, and then set it back to Low Latency.  If you do that it will eventually honor the setting.  You can check on the "interval" in mqtt for the device.  10Hz should be 100ms interval.

LuxStatera, 2022-12-31:
I gave that a go but it is still super slow. I'm not sure what is wrong unfortunately

DTTerastar, 2023-01-01:
What interval do you see (it's a property of the device in MQTT Explorer)

LuxStatera, 2023-01-02:
I only see the amount of messages and all of the other devices have 10x the amount of messages my Android does. Just by watching I can see the messages of the other devices change every second or so while my Android has not changed for a long time. I assume it has not changed in 1 hour because it shows on Home Assistant that the state was updated 1 hour ago

closed 2023-04-16 by stale[bot]