ESPresense/ESPresense #780: Restart every X hours or go into low power at certain times

khrisperry, 2022-12-31:
### Is your feature request related to a problem? Please describe.

Hello, I am using ESPresence in my home as well as outside to track my dogs.  We live on 5 acres so I have these placed around my property, but I can't run dedicated power to them.  So I use solar with a couple batteries.  The problem is depending on solar for the day, sometimes they don't have enough juice to make it to morning (mostly during winter)  When this happens the ESP32 ends up in a funky state.  It is still powered on, but no longer connected to the wifi.  So if I hit the reset button it pops right back online.  I was hoping someone could add a feature to allow the device to automatically restart say every 6 hours.  Another helpful feature would be to allow the device to go into low power mode from 10 PM to 5:30 AM or something like that to reduce the battery power at night.   

I'm not technically astute enough to figure out how to do this, but I'm hoping someone else might be.  Thanks for any help someone might be able to provide with this.  

PS.  I love ESPresence, we us it all over the house and property!!  Thank you!

DTTerastar, 2023-01-01:
Do you have the battery hooked up to an analog pin?  We could add support to throttle if the battery is less than a certain amount.  The macchina flavor has an example of a 12v batt hooked up to a analog pin w/ a voltage divider.

khrisperry, 2023-01-01:
I actually have the battery connected to a solar controller and the solar
controller is then connected to the USB port on the ESP.

closed 2023-04-15 by stale[bot]