ESPresense/ESPresense #790: No connection with underscore in SSID

dystechnic, 2023-01-20:
### Describe the bug

My IOT/Domotica SSID has an underscore in the name. I.E. " my_ssid" ESPresence will not connect to it. It keeps trying for a while but at the end it starts its local AP.
`Connecting to WiFi SSID 'my_ssid'...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................`

For testing purposes I created a new SSID without an underscore with exactly the same settings and security and the connection succeeds instantaniously. 
`20:29:46.751 > Setup network
20:29:47.255 > Connecting to WiFi SSID 'test'.....................................................192.168.30.163
20:29:50.019 > Firmware:     esp32
20:29:50.021 > Version:      v3.1.1
20:29:50.023 > WiFi BSSID:   D8:C7:C8:44:B8:64 (channel=11 rssi=-54)
20:29:50.027 > IP address:   192.168.30.163
20:29:50.030 > DNS address:  192.168.66.10
20:29:50.032 > Hostname:     espresense-test2
20:29:50.036 > Room:         test2
20:29:50.037 > Mqtt server:  xx.xx.xx.xx:1883
20:29:50.040 > Max Distance: 16.00
20:29:50.042 > Init Free Mem:138296`

Renaming the SSID is not a real option for meconsidering the amount of devices on that network.

### Version

3.1.1

### Device (be specific)

ESP-32s

### Logs

```text
Connecting to WiFi SSID 'my_ssid'...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
```

knewg, 2023-02-21:
I seem to have run into this problem as well. My Wifi is called **Secret IoT** (i.e. a space in the name), and I'm experiencing the same issue

DTTerastar, 2023-05-25:
Can anyone confirm this is an issue?

iMike78, 2023-05-26:
> Can anyone confirm this is an issue?

Not for me. I have an SSID with underscore, and 4 espresence node connected to it with no problem.

dystechnic, 2023-05-26:
Forme it still is.
That is, it it not a consitent issue. Sometimes when one device cannot connect another can. But the next time that one tries to connect is cannot. From that moment on it will never connect again.Only reflashing fixes the issues but again, not always.
I worked arround it by simplyfying the SSID ans password. So itmight be the combination of SSID and the passkey?

closed 2023-09-19 by stale[bot]

mn-box, 2024-12-14:
I just want to confirm that this issue still exists. I only found out today that this has been causing me a lot of pain of the past months. It has been very difficult to understand the source as connection is set-up and ping to the device works but further communication is not possible. What I also noticed is when this happens the vendor is not made available to my unifi network. Solution is removing the underscore in my SSID, but with over 50 wifi devices in my house that is quite a pain.